prusa3d/Prusa-Firmware #2514: [BUG][3.9.x] Significant E-axis behaviour difference (breaks prints) from 3.8.1

vintagepc, 2020-02-29:
**Printer type** - MK3S
**Printer firmware version**-  3.9.0 RC / 3.8.1

**MMU Upgrade** - MMU2S
**MMU upgrade firmware version** - 1.0.6

**Describe the bug**
There appear to be print-breaking E-axis differences with certain gcode combinations that were not present in 3.8.1.

The attached g-code prints fine in 3.8.1, but does not print properly in 3.9.0-RC. The perimeter prints fine, but any retraction over-retracts and the rest of the print continues without extruding until the over-retraction gets to the point it trips the filament sensor.

( for reference, the attached g-code is a first-layer speed and height calibration generated from 3dOptimizer.com  - it appears to be functional for many people on many makes of printers.)
  
**To Reproduce**
Print the attached g-code on 3.9.0-RC
  
**Expected behavior**
Gcode should print as expected, e.g. as it does in 3.8.1
  
**G-code**
[First_layer.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/4269742/First_layer.gcode.zip)

**Video**
Happy to capture one if necessary; I just don't have the time at the moment to set up a camera/flash versions/etc.

leptun, 2020-02-29:
Things changed a bit with https://github.com/prusa3d/Prusa-Firmware/pull/2381
Before the change the following combination would result in wrong behavior:
```
G91 ; set ALL axis as relative
M83 ; set Extruder ONLY as absolute
G1 E1 ; should be an absolute move, but still relative
```
With that change the extruder can be absolute even if all the others are relative.
I've noticed some time ago when I tried to use absolute extrusion in PrusaSlicer that it simply did not work. It did something similar to what I experienced with your gcode.
Your problem is that your gcode is relying on `G90` and `G91` to not affect the Extruder relative-ness. That was true in previous firmware, but it was accidental and only worked in edge cases. The marlin documentation states clearly that `G90`/`G91` should also set the extruder and that `M82`/`M83` are simply overrides.
As a potential solutions you can do the following in the unretract snippet:
```
G90
G1 Z0.150000 ; notice the lack of "E0.000000"
G91
```
Or you can add `M83` after every `G90` in your file.

leptun, 2020-02-29:
@vintagepc Do you have any way of testing the same gcode in marlin 2.0?
I'd like to see if it does the same issue also there. It should.

vintagepc, 2020-02-29:
No, sorry. This is my only printer. I did look at that combo of commands when it initially didn't work, and it left me scratching my head how it was supposed to work, since it does indeed appear to move the E axis to (abs) 0.000 - so it's working "as directed" and "as I read it" in 3.9. Ordoinarily I'd have been fine with that, it just got really confusing when hearing various success reports from others that were able to get it to print, prompting me to go back to 3.8.1 to test. 

There's other weirdness in the generated files from that site; TBH it looks like they were made by someone that doesn't quite get g-code (like the G28 at the *end* of the print, for example)

PR #2526 (2020-03-26): Mk3 decouple XYZE relative modes
This PR splits G90/G91 from M92/M93
This fixes the change that came in the 3.9 prerelease versions which unfortunately caused incompatibility with some gcodes.
G90/G91 no longer affects the extruder relativity. This is in contrast to marlin 2.0, which handles these commands just like we did prior to this PR.
Gcodes that were tested and were confirmed to work:

- File sliced with the latest PrusaSlicer MK3 profile.
- Gcode that comes with the printer on the SD card.
- https://github.com/prusa3d/Prusa-Firmware/issues/2514

Fixes #2514 
Fixes #2516 
PFW-1103
closed 2020-03-01 by vintagepc