prusa3d/Prusa-Firmware #2518: [ENHANCEMENT] Fan is on while waiting for nozzle heating

espr14, 2020-03-02:
i3 MK3S FW 3.8.1

1. Printer is printing
2. Filament runs out
3. Printer moves to the corner and turns nozzle heating off
4. I press button to start heating
5. Front fan is still running which slows down the heating.

Solution: turn fan off during heating and then turn it back on.

ManuGithubSteam, 2020-04-19:
+1 
This would alsp speed up normal heating of the alu block not just at a filament change...

gudnimg, 2022-12-10:
For filament change this issue was fixed recently here https://github.com/prusa3d/Prusa-Firmware/pull/3746 (will be included in 3.13)

For resuming a print from RAM, this will be this will be fixed here https://github.com/prusa3d/Prusa-Firmware/pull/3443

espr14, 2023-08-01:
I sold my i3 so no way how to check that.

closed 2023-08-01 by espr14