prusa3d/Prusa-Firmware #2521: [BUG]<loading error>

fopeczek, 2020-03-03:
**Printer type** - MK3S
**Printer firmware version**-  3.8.1 

**MMU Upgrade** -MMU2S
**MMU upgrade firmware version 3.8.1 (1.0.6)

**The bug**
Filament loading does not work reliably all the time when using the "load to nozzle" function: Sometimes, during the final phase of the loading process I can hear loud clicking noise and the filemant that is extruded out from the nozzle looks somewhat cracked. 
  
**Expected behavior**
My intuition says that the nozzle is too cool to warm up the filament properly. OTOH when I manually force the extruder out using "Settings"->"Move axis"->"Extruder" I never experienced this problem.

**Video**
https://youtu.be/EwysyRh5uj4

leptun, 2020-03-03:
I think that your MMU is acting up. Looking at the video, it looks as if it barely detects the presence of filament at the gears and keeps pushing till it almost reaches the nozzle. After the filament is detected at the gears, the check routine kicks in which pushes the filament even further at a quite high speed, creating the grinding sound and extruding filament when it shouldn’t. It also appears that the loading fails in your case.
I’d check the chimney position and move it a bit to the right so that it triggers earlier on the filament and see if it still produces the grinding.

fopeczek, 2020-03-04:
"I’d check the chimney position and move it a bit to the right so that it triggers earlier on the filament and see if it still produces the grinding." 
How to do that? I checked many ways to do that and that fails.

fopeczek, 2020-03-04:
Also plese go to the https://github.com/prusa3d/Prusa-Firmware/issues/2524 bug. Because this is not so regular. OTOH that unenables me to print with MMU.

JohnnyDeer, 2021-02-04:
@fopeczek here is manual for IR calibration, wich should help as @leptun wrote. 
https://help.prusa3d.com/en/article/ir-filament-sensor-calibration-mmu2s_2245/

closed 2023-10-19 by github-actions[bot]