prusa3d/Prusa-Firmware #2523: [ENHANCEMENT]Rotating line

espr14, 2020-03-03:
i3 MK3S, FW 3.9.0 RC1

There is an animation during test providing feedback that something is happening. It is done by switching horizontal (size 5) and vertical line (size 8). This does not look good.

I propose this with higher speed:
[image]

vintagepc, 2020-03-03:
FWIW the reason for the way it is now is that it uses pre-existing characters already in the HD44780 character set; the displays only have a limited set of customizable characters available for use. (What you're proposing would require the addition of a new set of custom characters.)

leptun, 2020-03-03:
You can always use / and \, so that’s not a problem. I was thinking of implementing an efficient way of rendering these kind of animations with the lcd buffer. I might implement it later on.

closed 2023-10-19 by github-actions[bot]