prusa3d/Prusa-Firmware #2524: [BUG]/[FEATURE REQUEST] Filament sensor does not catch filament.

fopeczek, 2020-03-03:
**Printer type** - MK3S
**Printer firmware version**-  3.8.1

**MMU Upgrade** - MMU2S
**MMU upgrade firmware version 1.0.6

**Describe the bug**

Filament begins to load fine when I click "Load the nozzle". When it seems the filament has just loaded, the printer unloads it, and then tries to load it again, and the loop continues.

I believe the core problem is that the filament sensor does not register the filament presence.

When I manually push the lever mounted in "the chimney" when the filament looks loaded, filament loads fine.

**To Reproduce**
1. Assemble the Prusa MK3S with the MMU2S
2. Turn the printer on
3. Load the original Prusa PLA filament by using "Load to nozzle" function and wait.

**Expected behavior**
The filament stays loaded and is not ejected.

**Describe the solution you'd like**
I think that best solution would be to remake the chimney in such a way that the user can adjust the gap between the chimney and the lever. 

**Video**

The video contains the bug and the manual solution of pushing the lever mounted in the chimney

https://youtu.be/_pXHKgMzUj0

ljpaff, 2020-03-05:
Try loosening the rear screw that holds the detachable roller that pushes the filament and also try to loosen a little the screw that holds the whole assembly against the extruder body (the spring screw).

This problem usually occurs because the top of the assembly does not get enough inside the extruder body and therefore does not activate the sensor properly.

RolfHodl, 2020-03-17:
It also should have some protection against flickering of the sensor. So in some loading scenarios the sensor gets a sort 1 (sees the filament) and then when the gear is spinning it can produce a short 0 of the sensor, and this will trigger a reload. 

That should only occur when 0 is more then 2 seconds, I think that will safe some mmu2s problems with loading or to early abandon of the loading process to make it more stable.

JohnnyDeer, 2021-02-04:
@fopeczek, this issue have same reason as your issue #2521. I´m closing.

closed 2021-02-04 by JohnnyDeer