prusa3d/Prusa-Firmware #2525: [BUG]X&Y Axes sometimes slam when canceling a print

jhenley01, 2020-03-05:
**Printer type** - Prusa Mini
**Printer firmware version**-  4.0.3

**Describe the bug**
  When canceling a print the print the X&Y axes will sometimes slam against the ends very hard. It does not happen every time. It seems that when the cancel print option is taken the X&Y axes are set to move a certain amount so depending on how close to the front right the print head is it may or may not slam.
  
**To Reproduce**
  Start a print and cancel when the print head is near the front right.
  
**Expected behavior**
  The print heads should not slam.

leptun, 2020-03-05:
Where exactly does it slam? When it finishes the parking movement or when it stops instantly?
Sometimes if stop is selected during a fast movement the drivers will skip steps. Maybe that’s why the bed crashes the axis end.

jhenley01, 2020-03-05:
> Where exactly does it slam? When it finishes the parking movement or when it stops instantly?
> Sometimes if stop is selected during a fast movement the drivers will skip steps. Maybe that’s why the bed crashes the axis end.

It's during the parking movement after the Z axis lifts, it looks to me like after a cancel the X&Y axes are set to move forward and right a fixed amount and will slam if the print head was in the front right area when stopped. The machine homes before each print so I'd think it should know its position and how far it can move.

jhenley01, 2020-03-05:
I got a video of it happening.

[https://youtu.be/HSwy_sHWCMo ](https://youtu.be/HSwy_sHWCMo)

leptun, 2020-03-05:
I just realized that you have a Prusa MINI. This is not the repository for that printer. For any MINI issues, please post here: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues
There seems to be an already opened issue there. https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/257

jhenley01, 2020-03-05:
Ok thank you!

leptun, 2020-03-19:
@jhenley01 Please close this issue.

closed 2020-03-19 by jhenley01