prusa3d/Prusa-Firmware #2528: [BUG]<Linux Firmware Builder makes a mess of file permissions>

itcarsales, 2020-03-06:
**Printer type** - [MK3]
**Printer firmware version**-  [linux builder - MK3]

**Describe the bug**
  The Linux Firmware builder script uses sudo without need and creates a giant mess of root permissions.  It is not standard practice to create root owned folders and files in a users home directory.  Without specific directory structure, those unfamiliar with linux may find they have installed to locations they do not have access to...or cannot delete files created by the script.
  
**To Reproduce**
  Follow the existing installer instructions and look at the permissions of the files/directories created, and the need to execute future builds as root or with elevated permissions.
  
**Expected behavior**
  When installing to a users home directory, it is standard practice to make sure the user owns and can access/run/delete their own files.  Many users do not have or need root  access.  Simple file structure makes this easier to maintain. 

I am not a Linux guru, but I like a semi-clean workspace.....I rewrote the Environment Installer and the Firmware Builder to be more permission and user friendly.  Its all yours if you want it.  Thanks for an awesome printer!!

https://github.com/itcarsales/PrusaBuilder-MK3

leptun, 2020-03-06:
https://github.com/prusa3d/Prusa-Firmware/pull/2488
I think that this PR is related to your issue. You shouldn't need to use sudo to compile. That PR just has updated instructions.

itcarsales, 2020-03-06:
please try to run it without sudo to confirm.  I failed on two different flavors of debian due to permissions of files within the arduino zip folder.  This occured during the build process after the verify in all tests.

itcarsales, 2020-03-06:
Also, the existing navigation creates files/directories above the repo location which is dangerous in linux.  As root, as it is currently run, should be changed ASAP.  But even with user rights, this could cause a failure if the user keeps things in /home/user/ like so many do.  /home/ is not a writable folder with a standard user permission set.

leptun, 2020-03-06:
Ok. So I tried the following:
Clean Ubuntu 18.04.4 install.
Updated everything.
Installed git and python.
`cd ~/Downloads && git clone https://github.com/prusa3d/Prusa-Firmware.git && cd Prusa-Firmware && ./build.sh`
After this command that set up everything and ran build.sh, the firmware was compiled successfully without the use of sudo (for compilation). Files were created in `Downloads`, but nothing outside that. Unless you put the cloned repo directly in the user's folder, I can't see how the script would generate files in `/home`.
I might be wrong about this issue, of course, but other than removing sudo from the instructions there is nothing more we should do.
Also, if you want a more user friendly compile script, maybe you might be interested in `PF-build.sh`.

itcarsales, 2020-03-06:
I can duplicate your method, so if you would like to consider it closed, I will close the ticket.  However, I strongly feel this is not the correct way to build, nor create dierectory structure.  Not everyone uses vanilla ubuntu, many newer users following guides come from Raspbian where it is common practice to dump files firectly into the user directory.

I have also not completely ruled out the initial install with sudo causing problems in later attempts.  I thought I had, but I need to look at my images later to see if I had arduino environments or not. It was late.

I think I will continue with my repo to create an alternative regardless.

Thank you for taking the time to respond and to assist! It is most appreciated

closed 2020-03-06 by itcarsales

leptun, 2020-03-06:
The building process is quite messy and it’s always been this way. However, recently things got a bit more organised, especially in the build environment area.
As for the instructions, they still are a mess for build.sh, but PF-build.sh has much better documentation.
So what are your suggestions in the end? Removing the need for sudo in the instructions and keeping the firmware build environment inside the cloned folder (+.gitignore).

itcarsales, 2020-03-06:
That would be an excellent start, yes.....a step further would be to recommend or require a specific, universal location like /home/user/HERE to allow easier location references.  I would also suggest setting permissions explicitly on anything that has been unpacked in the process, but that is not as pressing.  If the build process requires a particular Linux flavor, or at least has a known working list, that would also be helpful.  One thing I have yet to figure out is the effect this had on my existing /home/user/.arduino15 directory and contents.  I had to remove them manually to reload the most recent IDE after working with the older version.

My plans are to keep this confined to a directory where I can manage permissions independently, and maintain isolation until I learn far more about it.  I will look into PF-build.sh tonight, thanks.

itcarsales, 2020-03-06:
My final suggestion on this topic would be to keep the "Environment Setup" separate from the "Firmware Build".  Separating the two allows for easier updating of the individual parts, and a clearer understanding of what each is trying to accomplish.  It would allow for easier maintenance and future updates......but that is more of a personal opinion that fact