prusa3d/Prusa-Firmware #2533: [BUG]3.9.0-RC1 PowerPanic Failure on SD-Print started from Octoprint

TheZeroBeast, 2020-03-08:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

MK3S on 3.9.0-RC1

MMU2S but disconnected for this issue

**Describe the bug**
When an SD file is started via Ocoprint interface the EEPROM_FILENAME, EEPROM_DIR_DEPTH & EEPROM_DIRS memory locations don't get updated as they do when starting SD print via LCD Menu.

This causes issues when Power Panic is used as the recovery tries to open the previously filename that was started via LCD Menu.
  
**To Reproduce**
Start SD print via a connected Octoprint instance
  
**Expected behavior**
same as via LCD Menu when starting print so power panic works as intended.
  
**G-code**
M23 is where the action is at I believe

leptun, 2020-03-08:
I am working on an updated Power Panic save/restore implementation that will save the filename and dirs at the moment of Power Panic, not when the print is started. This should in theory fix the issue.
If everything goes well, this will be ready for 3.10.

TheZeroBeast, 2020-03-08:
Yes, that would solve the issue.
At the moment I'm testing saving the values in M23 code which appears like it's resolved it.

leptun, 2020-03-08:
If you want to take a look at my new implementation, here it is: https://github.com/prusa3d/Prusa-Firmware/compare/MK3...leptun:MK3_XVLO
The idea behind it is that the EEPROM is very slow and it wastes a lot of time saving (or more precisely erasing) the data. That’s why I implemented a PowerPanic version which uses the XFLASH (the SPI 2mb chip that holds the language data on the MK3/S) to store a lot more state variables during UVLO. It is waaaay faster than the EEPROM.

leptun, 2020-03-08:
Btw, M23 is used in many places, some even a bit unexpected. Make sure you don’t overwrite the file during PowerPanic recovery or when the file is checked if it’s complete.

TheZeroBeast, 2020-03-08:
Awesome, that sounds like a nice solution. I'll have a good look.
Yes, it does get called a bit.
My test only wrote to EEPROM when not sdprinting so it only records as the sdprint is starting

leptun, 2020-03-08:
To start a print, all you need is these two commands.
M23 to select the file and open it.
M24 to actually start printing the open file. Sometimes also acts like resume.

TheZeroBeast, 2020-03-08:
In your XFLASH codebase above where does the filename/PowerPanic variables get saved?
I can't see where filename for PowerPanic is stored in both M23 & the LCD Start SD Print implementations

leptun, 2020-03-08:
It is not saved in neither of them. In my implementation, I have the luxury of doing everything during the actual Power Panic and don't need to worry about saving things in advance.
```
    // Write file information
    {
        uint8_t depth = (uint8_t)card.getWorkDirDepth();
        XFLASH_WRITE(depth, XVLO_DIR_DEPTH);
        for (uint_least8_t i = 0; i < depth; i++) {
            XFLASH_WRITE_WITH_SIZE(dir_names[i], XVLO_DIRS_ARRAY + i * 8, 8);
        }
        // Write the full filename filename (name+extension+NULL)
        XFLASH_WRITE(card.filename, XVLO_FILENAME);
        // Write the file position.
        XFLASH_WRITE(sd_position, XVLO_FILE_POSITION);
    }
```

TheZeroBeast, 2020-03-08:
Ah, wonderful, that's much cleaner.

leptun, 2020-03-09:
What's your opinion on my XVLO implementation? Do you see any way I could improve it?
Right now, the biggest disadvantage is that you must be sure that the flash is initialized and ready for a Power Panic. If the data is not erased from the flash prior to the Power Panic, then saving will fail.

TheZeroBeast, 2020-03-09:
It sounds solid.
If that is the case then on firmware flash I would prep XFLASH addresses ready for a PowerPanic.
When recovering from a PowerPanic erase addresses again in preparation for next event.

It shouldn't be effected if I build my hex myself and without the extra languages I would guess 🤔

leptun, 2020-03-09:
I think it works also without extra languages, the only downside being that you can't read the contents back for debugging since the second bootloader is not present without extra languages. I only used that function when I was debugging saving since I could dump a .bin of the whole flash through `avrdude-slic3r`.
I save all UVLO variables to XFLASH with the exception of two:
- EEPROM_UVLO - determines if Power Panic is triggered or not
- EEPROM_TINY_Z or something like that - saves Z position after being raised. Also handles a UVLO during recovering from another UVLO.

These two must be saved to the eeprom since we only erase the XFLASH XVLO sector after a full recovery. Not only that we can't load everything at the beginning, but there's also the problem of the fact that erasing the XFLASH is very slow compared to writing to it, so there would be about 300ms where if Power Panic would occur, the data would be lost.

Btw, if you want `avrdude-slic3r`, I can post the binaries. It is the one included with PrusaSlicer and it is the only software that has support for dumping the XFLASH.

TheZeroBeast, 2020-03-09:
Very cool if it works without the other boot loader hassle.

If you have a link that would be great thanks @leptun 😊

leptun, 2020-03-09:
[avrdude-slic3r.zip](https://github.com/prusa3d/Prusa-Firmware/files/4305623/avrdude.zip)
Here you go. There you also have some scripts that can help you for automating the flashing/dumping tasks.
I compiled for WIN-64 and linux-amd64. You can compile this by yourself from the PrusaSlicer source, but why go through the hassle when you can use these precompiled binaries.
Please note that I only tested the windows variant. The other one should work in theory, but I never actually used it as a tool.

TheZeroBeast, 2021-02-03:
Hi @leptun, do you know if this issues resolution has been progressed?

leptun, 2021-02-04:
I'm working on something related to this now. I'll probably have this fixed this week if everything goes well.

leptun, 2021-02-04:
It seems I've only been able to partly solve the issue. I'll elaborate on this some other time.

closed 2023-10-19 by github-actions[bot]