prusa3d/Prusa-Firmware #2535: [FEATURE REQUEST] Use same heat cycle as Prusa MINI

PotatoFi, 2020-03-11:
**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]

**Is your feature request related to a problem? Please describe.**
  Heating both the bed and nozzle to operating temperatures causes filament to ooze and stick to the bed during the autolevel sequence. The Prusa MINI preheats the bed, but not the hotend; PETG for example only heats the nozzle to 170 °C until the autolevel sequence has completed, which completely fixes the oozing problem.
  
**Describe the solution you'd like**
  Before autolevel, heat the bed to operational temperature, and only heat the hotend to 170 °C. After autolevel is complete, bring the hotend to operational temperature as the very last step before beginning the print. Basically, implement the exact same behavior as the MINI on the MK3/MK3S and similar printers.

PotatoFi, 2020-03-11:
Immediately after posting this, I realized that this is handled in PrusaSlicer, NOT firmware. Attempting to close this issue since this is not the appropriate project for it.

closed 2020-03-11 by PotatoFi

leptun, 2020-03-11:
I was just about to write the same thing. :)