prusa3d/Prusa-Firmware #2536: 3.8.1 firmware on i3 MK3S

JohnFielden, 2020-03-12:
I just installed the latest firmware on my i3 MK3.

The cal wizard ran until I got the XYZ calibration.  Then it fails when looking for the first of 4 probe points.

Auto home works ok.

When I try to rerun the xyz, it fails the same way everytime.

My printer was working perfectly prior to the firmware update.

JohnFielden, 2020-03-12:
The error occurs in the 2nd step.  After the extruder rises up to 210, it tries to find the probe points.  Warns me to put paper down then goes looking, but it is coming down off of home.

When I auto home it goes to the right place.  It seems the firmware's endstop check puts the platter off, then the z check doesn't move to home first.  Bug?

leptun, 2020-03-12:
Can you please post a video wit the issue? It would help a lot.
Did you flash the MK3S firmware instead of the MK3 or the other way around?

Prusa-Support, 2022-05-14:
Hello! Sorry for intervening on this thread so late. I am closing the issue for now, I assume it has been solved - in case there are still issues related with this, can you please let me know via email? info@prusa3d.com mentioning me in the subject ("attn: Alessandro"), we'll look into it in more detail in that case. 

Alessandro Pantaleo
Prusa Research

closed 2022-05-14 by Prusa-Support