prusa3d/Prusa-Firmware #2543: [BUG] Frimware 3.9.0-RC1 print quality degration compared to 3.8.1

nikkolade, 2020-03-16:
**Printer type** - MK3S
**Printer firmware version**- 3.9.0-RC1
**MMU Upgrade** - MMU2S
**MMU upgrade firmware version 1.0.6

**Describe the bug**

Print source: https://www.thingiverse.com/thing:3494496
and explicitly: https://www.thingiverse.com/download:6386081

While printing the above STL sliced with PrusaSlicer 2.20-RC2 and using PETG material works just fine when using the MK3S firmware 3.8.1 (build result on the left), the very same gcode results an awful quality after the printer was upgraded to 3.9.0-RC1 (end result on the right).
Please, see the attached picture.

  
**To Reproduce**
Upgrade firmware MK3S firmware to 3.9.0-RC1
and print the above mentioned STL file.
  
**Expected behavior**
The print quality should not degrade, compared to previous firmware version.

 

[image]

wavexx, 2020-03-16:
Could you post the 3mf file saved by PrusaSlicer using the same settings you used to print it? Also posting the gcode itself would help so we can have a look.

nikkolade, 2020-03-16:
Sure, here is the 3mf file.

[3mf.zip](https://github.com/prusa3d/Prusa-Firmware/files/4339149/3mf.zip)

nikkolade, 2020-03-16:
And the gcode
[gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/4339156/gcode.zip)

vintagepc, 2020-03-16:
Hm... have you tried re-calibrating your linear advance K-value with 3.9.0? I'm not saying the default shouldn't work fine, but I'm curious whether calibrating for the 1.5 K-value makes the issue go away.

nikkolade, 2020-03-17:
No, I haven't. What is the proper way to run the linear advance calibration ?

leptun, 2020-03-17:
You can find instructions on what Linear advance is and how to calibrate it here: https://marlinfw.org/docs/features/lin_advance.html
You are looking for LA 1.5.

vintagepc, 2020-03-17:
FWIW, I hadn't paid it much mind but since you are also using PETG... I've noticed my PETG prints are also a little sloppier in terms of strings, sags and zits, but nowhere near the picture levels of bad. I initially ascribed it to wet filament but a fresh roll didn't show any improvement, so there may be something to this issue. I'll also try running an LA1.5 calibration for PETG next time I load it up. 
PLA's been fine so far.

DRracer, 2020-03-18:
@nikkolade thanks for reporting, this is a serious issue and we must solve it before the 3.9 final goes out.
Could you please do the LA15 calibration test as @leptun proposed? We are especially interested in the "correct" K-factor you get (a photo of the whole print sheet with calibration lines will be great).

nikkolade, 2020-03-18:
[calibration.zip](https://github.com/prusa3d/Prusa-Firmware/files/4350106/calibration.zip)

I don't know what is the problem with this particular calibration, as I only increased the temp to 210C and changed some filenames, but for some reason the included gcode crashes the printer right after the initial bed calibration. Therefore I was unable to print anything with this test.
I tried a few times from octoprint and from SD-card, didn't matter at all.

vintagepc, 2020-03-18:
I used that code/page recently for flex, so the gcode should be okay. 

I see you have an MMU. There's a known bug if you try to extrude with IR=0 and FINDA=1, or a runout detected with a "?" on the screen for the filament. (Fix is pending merge)

Try doing a load to nozzle first if you haven't already done so - the gcode does not have the MMU single mode Tx/Tc commands that prompt filament selection.

leptun, 2020-03-18:
@nikkolade Just tried the gcode on a MK2.5S without MMU and there was no crash. It is as @vintagepc said.

nikkolade, 2020-03-18:
Yes, manually loading to nozzle first allowed the test to be executed.
Of course the PLA-filament I used is clear, so the end result is slightly difficult to see. I can certainly run it again with for example white PLA or some PETG if needed.
[image]

leptun, 2020-03-18:
The calibration is technically filament specific, so you might want to use the orange PETG for the calibration.

vintagepc, 2020-03-18:
Also, you're going to want to run a lot narrower range. Your 0.1 line is already showing overextrusion at the start of the fast section, and it just gets worse. Try something like 0 to 0.2 with 0.02 steps instead. 
What you're looking at here are the transitions from slow to fast and back again - ideally you should have some that are too fat at the start of the fast segment, and some that are to fat at the end, to show you've got some values on either side of the correct one. 
You want the most uniform line you can get, it should not be possible to see where the fast travel part starts and ends. 

Hope that helps clarify what you're looking at. You might need a few cycles to narrow down the right range of values to test.

nikkolade, 2020-03-18:
Ok, yet another test result. Any tuning suggestions after this ?

[image]

DRracer, 2020-03-18:
@nikkolade looking at your calibration print I'd say the best K factor is 0.08. S far we've been using 0.12 AFAIK.
Now please take your g-code and search for:
M900 K45 ; Filament gcode
Change it to:
M900 K0.08

Save and run a few centimeters of your container print to see if there is any difference.
We'll be doing similar tests at Prusa HQ tomorrow.

vintagepc, 2020-03-18:
.08 looks pretty good to me - Edit your custom filament g-code for the orange PETG and change the K value for the M900 command, e.g. `M900 K0.08`

Then try your original print again (be careful you don't revert the K change if you re-open the 3MF file!)

You can probably skip doing the whole print and just cut out a chunk of it or stop a few cm in where the issue would have manifested itself

nikkolade, 2020-03-18:
Ok, I'll let the printer try again with the modified gcode. Let's see if the end result is equal to FW 3.8.1 this time. I'll let you know latest in the morning.

Panayiotis-git, 2020-03-18:
I would definitely use a smooth PEI sheet to determine the K factor. :smile:

nikkolade, 2020-03-18:
Ok, the K0.08 value modification improved the end result a lot!
[image]

vintagepc, 2020-03-18:
Interesting, and good to know. 
I'm guessing that's Prusament PETG?

I'll spot check my Overture (AKA former Amazon) PETG tonight and see if I get a similar value or if it's closer to 0.12.

DRracer, 2020-03-18:
@nikkolade interesting result, please let the print finish if possible for detailed comparison with the one printed with FW 3.8.1
Btw. how old is your nozzle - i.e. how much filament did go through (roughly)?
@vintagepc please do a LA15 calibration on your machine too
@wavexx @leptun FYI

vintagepc, 2020-03-18:
Will do. Current print is finishing in about 20 minutes. Will post pictures of the plate.

vintagepc, 2020-03-18:
Results are in. For me, PLA @ 210C (red) comes in best at 0.04, PETG@235 (black) also at 0.08.

[image]
[image]

I do have a Skelestruder but the gear->nozzle distance isn't as different from stock as it used to be; the 3S revision shortened that a bit; so my values might be a smidgen lower than expected. (FWIW I was running 20 for PLA and 35 for PETG in LA1.0, but those were not calibrated at all and just going off a generic comment that I could drop the Mk3 values by ~10. )

nikkolade, 2020-03-19:
The print is now complete and there seems to be still a big difference with the latest 3.9.0-RC1 and the older 3.8.1 versions. The print looks much better from the front after the K-value change.
However, the lefthand side of the print is still much more worse than the other sides.

The filament is Prusa PETG, not prusament. All my filaments have been dried every once in a while, so they are not wet. The printer itself is originally of MK3 model and it has been delivered on December 2018. It has not been used heavily though as there has been occasionally months between the prints. I quess 3-4 complete rolls and half a dozen non-empty rolls.

[image]
.
[image]

DRracer, 2020-03-19:
@nikkolade oh, this is interesting - it looks like some error is being accumulated after those many retractions.
@wavexx what do you think?

vintagepc, 2020-03-19:
I don't think that's K value related anymore. The parts that are not clean (esp. right under the thicker bands) all have a very "melty" look to them, like there is not enough cooling. Notice how it "shrinks" inwards instead of just being extra ooze attached to the exterior of the part like previously. 
This seems even more likely given the front of the print is reasonable and the sides have issues

@nikkolade Has your filament been sitting out in the open for a while? PETG absorbs moisture and that can produce symptoms like we are still seeing. How much time elapsed between the print you made on 3.8.1 and the reprint with the RC firmware?

nikkolade, 2020-03-19:
I did the two prints after each other, ~8h difference. The latter one was using the FW 3.8.1 and the same roll of PETG, so the FW3.8.1 print should have had the quality issues, if it would be caused by moisture in the filament.

I have a food drier and this particular roll has been dried 2 days ago. After drying, the filanents are kept in a ikea samla plastic box that has been "sealed", together with silica gel pouches (a lot of them, also recently dried).

I bet that I can print again one of those containers with the older 3.8.1 and using the same filament roll and they will still look good.

leptun, 2020-03-19:
@wavexx Does LA1.5 slow down acceleration even when just Z and E moves are done? Maybe these are slower with LA1.5 compared to LA1.0 which didn't take Ejerk into account.

vintagepc, 2020-03-19:
I'll give part of the print a go today on my own setup to see whether I also have the same artifacts even with a corrected K-value. It won't be completely conclusive but very informative as to whether it's heat related.

vintagepc, 2020-03-19:
Print still in progress, but quality is indeed abysmal. Part of the problem is the thin vertical sections flex quite a bit, which also results in uneven sides and stringiness. Will edit in a picture when it finishes. I am experimenting with tuning cooling/temps to see if I can get an improvement.

vintagepc, 2020-03-19:
Update:
I think this is indeed compounded by a cooling issue. Not sure why that would have changed from 3.8 to 3.9 though.

I printed a shorter part but the results are similar.  It looks like the worst parts of the print for me are those that are shielded somewhat by other print parts, whereas the best ones are most exposed to the cooling fan airstream.
Part as printed on the bed:

[image]

I made tuning adjustments as the print progressed:
-> No changes until halfway up the first course of spindles
-> Dialled back temp to 230 (from 235), increased fan setting to 110 (from 76) halfway up the first stage
-> For the second stage of spindles, I dialled up the fan to 200, flow to 95%

Picture of the cooling duct:

[image]

Additional part photos:
[image]
[image]
[image]

wavexx, 2020-03-19:
It always looked like a cooling issue to me more than LA itself, but given a higher K factor will lead to slower jerk the issues are probably connected. These thin features will overheat quickly.

The gcode is resliced already at E-jerk=4.5, which looks good. Did you change it yourself?
I also see Tx/Tc gcodes which I didn't see before.

To respond also to @leptun, any move which is not involving extrusion is not changed by LA (so wipes/jumps shouldn't be affected).

For the stock Prusament Orange I do use K=0.12 at the default temperature of 250C.
For the "Prusa clear PETG" which I use at 240C ("generic pet" profile) I've set up a K=0.13 even. It depends a bit.

I'd love to try this, but I'm currently restricted in movement for this covid19 thing, so huuh, it's a bit hard to test stuff. I don't have PETG at the moment :/

I'd try two thing out of my mind:

- reduce retraction length: the default is 0.8, I'd go for 0.5. I know it sounds counter-intuitive, but if the LA factor is properly tuned there's generally a lot less pressure left at the end of the extrusion. Some retraction is still needed to avoid ooze, but reducing it will reduce the time spent on small features like these.

- bump E-jerk even higher. In machine-limits, in "maximum jerk E" try to increase to something much higher, like 8. This is not a solution in itself, but it will increase acceleration overall which is useful to see if results in an appreciable difference.

vintagepc, 2020-03-19:
> reduce retraction length: the default is 0.8, I'd go for 0.5. I know it sounds counter-intuitive, but if the LA factor is properly tuned there's generally a lot less pressure left at the end of the [...]

FWIW my default retraction is 0.4.  I'll up the E-jerk in the .gcode file and reprint to see what happens.

Kachidoki2807, 2020-03-19:
Not sure how it can help, but I just gave a try to the K-factor calibration with Stratasys Blue ABS-M30. I got the best result at 0.04 on a MK3s.

nikkolade, 2020-03-19:
As I happened to have some use for an additional silica gel container, I printed with the same original PETG filament roll & FW 3.8.1 the following items (75mm size this time) and the end result is very good in my opinion. There is certainly some tuning to be done for the upcoming FW version before it will produce the same quality out of the box.
[image]
[image]

vintagepc, 2020-03-19:
No improvement changing the E-jerk to 8. 

I had a go at trying to back out LA1.5 from the current RC to confirm if it is the origin of the difference. Alas this introduces a few compile errors I'm not easily able to resolve without slogging through the history of those changes.

wavexx, 2020-03-19:
On Thu, Mar 19 2020, vintagepc wrote:
> I had a go at trying to back out LA1.5 from the current RC to confirm
> if it is the origin of the difference. Alas this introduces a few
> compile errors I'm not easily able to resolve without slogging through
> the history of those changes.

You can set K0 to disable LA completely. This will generate some
overextrusion on the corners, but this would help determining if LA can
be the source of the issue

vintagepc, 2020-03-19:
Well, the simplest solution... :tada: 

Anyway, print in progress. Looking good so far...

vintagepc, 2020-03-19:
Well, I think these pictures speak for themselves. (I cut off just a segment to speed up the testing)

[image]
[image]

wavexx, 2020-03-19:
Is this comparable to 3.8.1 all-else being equal?

vintagepc, 2020-03-19:
I don't have a 3.8.1 reference to compare it to but it looks very much like the OP's 3.8.1 print in terms of stringing/quality. Slicing settings are identical to the lower half of my other print; only difference here is M900 K0 instead of K0.08

If it's beneficial I can compile 3.8.1 for my printer and try the same g-code tomorrow. (Can't use official release as 3.8.1 doesn't have #2306 and my setup seemed to hit that last time I flashed back.)

wavexx, 2020-03-19:
I'm going to try to reproduce the issue (or at least try to see if I can notice a difference) using regular PLA. At this point it's either still too much e-jerk influence or there is some odd overextrusion going on that hopefully I should be able to spot with PLA

vintagepc, 2020-03-19:
That sounds quite possible - Observing the previous prints most of the stringing and junk is from PETG that has crept up the sides of the nozzle and is being dragged around partially sticking to fresh bits that get laid down.

wavexx, 2020-03-19:
Indeed, once some petg starts sticking to the sides of the nozzle it's going to stick it everywhere until a long extrusion cleans it all up (in this case the rings)

vintagepc, 2020-03-20:
Yeah, and that's with a Ni-plated nozzle too, so it strings off in chunks more easily than brass. 

Overextrusion certainly explains the meltiness, since if you're adding extra plastic, you also add more heat that needs removing - which is why cranking up the fan helped quite a bit. The test part wasn't flexing nearly as much while the spindles were printing.

leptun, 2020-03-20:
In the early days of the LA1.5 for the Mk3 I also notices some cases where overextrusion would occur. It’s mostly with really small segments.
The place I encountered it was with some infill that was filling a small gap between perimeters on the first layer.

I also think that the “corners” of a solid infill (when it finishes and makes smaller segments) got worse with LA1.5.

leptun, 2020-03-20:
@wavexx Was the K0 compatibility issue fixed or does it still lock to one value or another? It would be best if all these tests were done without this kind of issue since it would most likely create false results with calibrations.

wavexx, 2020-03-20:
> @wavexx Was the K0 compatibility issue fixed or does it still lock to one value or another? It would be best if all these tests were done without this kind of issue since it would most likely create false results with calibrations.

Is there an explicit # for this issue? I opted to use K-1 as a manual "reset". The reset is still done between SD prints, and I would love to implement "print timers" to have an extra hook to reset stuff between prints when using a host (see https://github.com/prusa3d/PrusaSlicer/issues/3724)

K0 still disables LA, but without the mode reset. The reason is that slicer++ (supermerill's fork) allows to customize the K factor by fill-type, so for example you can decide to disable LA during infill. Resetting the mode on K0 could introduce random swaps during the print, which I think is worse.

leptun, 2020-03-20:
If the gcode starts with a K0 and then swithches to another value, will the compatibility be locked because of K0 or will it switch accordingly when K0.01 (LA1.5) or K5 (LA1.0) are seen later?

wavexx, 2020-03-20:
Mode is decided on the first non-zero value seen. I've been thinking a lot whether adding an explicit "set mode" flag would be a good idea or not, given marlin has none :/

DRracer, 2020-03-25:
@nikkolade just a short notice - interestingly, we haven't been able to reproduce you issue so far, we have multiple prints from PETG coming out beautifully in FW 3.8.1 and 3.9.0-RC1, i.e. LA10 and LA15 - directly from your G-code.
I'm not saying there is no issue (there obviously is at least on your and @vintagepc 's printer), we are still searching. Any even minor notice/comment may help.

nikkolade, 2020-03-25:
Ok, this is baffling new to me. Could there be something in my old printer that is different compared to your's ? I just checked and my printer has been delivered on Dec 2017 as an (early) MK3. It has been updated later on to MK3S and I have added the MMU2S unit at the same time.

I really don't understand, how otherwise my printer could have issues with the 3.9.0-RC1 FW, while your test machines work fine. After all, with the FW 3.8.1, my printer works just nicely and produces good results. Has there been some updates in the HW afterwards, that make the newer models print with better results ... ?

I assume that you have used a similar printer with a textured sheet in at least some of the tests ... ?

vintagepc, 2020-03-25:
Interesting. My own printer was received early 2019, with a Mk3S & MMU2S update as well. 

I will see if I can do some more experimenting today to try to identify what is going on.

@nikkolade I'm guessing you have a completely stock printer? If yes, that likely eliminates any of the modifications on mine as interferents.

Can you weigh both of your prints to see if there's an appreciable difference in weight between the good one and the bad one? That might help narrow down whether it's an overextrusion issue or primarily a heat issue.

nikkolade, 2020-03-25:
Yes, my printer is completely stock. I did add the updated MK3S (latest github models, compared to originals from 2017) Y-belt printed parts a few weeks ago though.

vintagepc, 2020-03-25:
OK. To give a complete rundown on my mods:
-> 0.9 degree steppers on X/Y
-> Skelestruder (3.5:1 gear ratio). 
-> Firmware is stock + #2526 + #2499 + #2508 + #2473 + #2464 + #2201 (basically all my own open PRs except for the first one)
There are also a few minor load distance tweaks to correct the MMU behaviour with the skelestruder, but nothing that impacts anything other than load/unload.

vintagepc, 2020-03-25:
I think I've just confirmed that this is definitely overextrusion:
- I sliced a small part of the offending model that takes about 30 min to print.
- Sliced G-code.
- Edited code to change to K0, saved as a separate file. All other contents identical.
- I cut two identical 1m lengths of filament.
- Used the first to print with M900 K0.  166 mm of filament remaining after the print completed.
- Loaded the second and printed the file with M900 K0.08. 54 mm of filament left after it completed. 

So enabling linear advance pushed an extra 112mm of filament (>10%, or, 269mm^3) of plastic into the model. I'd say that is well outside the measurement error which I'd expect to be give or take 10mm (the remaining plastic in the melt zone from load/unload between prints)

vintagepc, 2020-03-25:
Further update - I ended up getting a *good* print with a different speed print profile with LA1.5 enabled. It was probably still overextruded (did not do the filament length test on it, can do so if desired) but I was printing perimeters at a much faster rate and at 0.2mm layer height. Top left in the images. The spindles look wider than the "correct" ones. 

Reversing the overextrusion calculation and running the "bad" print gcode with a flow rate of 88% produced a good print too. (Bottom left)

I think that means that this particular combo of part and print settings just makes the issue show up particularly well, owing to solid geometry, lower layer height, and a material that stays soft and gummy quite a while.

(Top row is various parameter tweaks based on the working LA slice I was experimenting with, middle row is the overextrusion test from the previous post, bottom row is the reprint at 88% flow)

[image]

[image]

vintagepc, 2020-03-26:
I think we have a winner - LA is not working properly on retractions/deretractions. 

Disabling retraction results in a reasonably good print, barring the sort of stringing you'd expect from such a setting - but not the overextruded melty mess we are observing in other cases.

I browsed through the code and I suspect there is cumulative error creeping in from the asymmetry of retractions and de-retractions, where negative extruder moves are not compensated, but forward moves are.
It appears to be constant and not related to retraction distance, my default is 0.4, and changing to 0.2 and 0.8 did not appear to make the result any better or worse than 0.4

DRracer, 2020-03-26:
@vintagepc exellent analysis, good job
We have been able to simulate similar results with an MK3S with "MMU" extruder. A "normal" machine doesn't do this.
Probably rising the E-jerk beyond limits (say 20) can also solve the problem, probably the retractions were fine-tuned in previous FW's with respect to the inaccurate material flow and mechanical properties of the extruder.

[44 more replies]

closed 2020-08-12 by JohnnyDeer