prusa3d/Prusa-Firmware #2544: [BUG]<Filament sensor error during "selftest" calibration, MK3S, FW 3.9.0RC1>

Tanischacat2009, 2020-03-19:
When calibrating my MK3S printer, I encountered an error message during self test regarding the filament sensor.

I normally have the filament sensor disabled because it doesn't always work.  A few months ago, I replaced the sensor with the updated version that Prusa has available in their eshop.  When I did the calibrations (with the sensor disabled), I got an error message saying that a "HIGH" value was expected, and the selftest halted.

I reset the printer, enabled the filament sensor, and tried this two more times...getting the same error message.  Previous firmware builds didn't have this issue.

Another small issue, is when I tried the belt test, I got an x-axis length error, this was despite all axis passing with my previous 3 self test attempts.

All other calibrations work as they are supposed to.

ochm, 2020-03-21:
I had the same problem with fsensor during selftest, but the sensor works well whent tested while printing.

mafredri, 2020-03-23:
I had the same issue, although my fsensor was already acting up on 3.8.1 (as well as on 3.9.1-RC1). Turns out the fsensor is too close, simply adding a 0.2mm (printed single layer) spacer under the sensor below the screw and another at the top seems to have done the trick.

Tanischacat2009, 2020-03-26:
> I had the same issue, although my fsensor was already acting up on 3.8.1 (as well as on 3.9.0-RC1). Turns out the fsensor is too close, simply adding a 0.2mm (printed single layer) spacer under the sensor below the screw and another at the top seems to have done the trick.
> 
> > Another small issue, is when I tried the belt test, I got an x-axis length error, this was despite all axis passing with my previous 3 self test attempts.
> 
> This also happened to me, but only because Z was maxed and the backside of the extruder hit the filament stand/holder.

In my case, the Z access was only a couple of centimeters above the build plate when I did the belt test.  Just found it odd to get an "X-axis length error" at that point when all axis passed self test 3 times in a row just previous to that.

closed 2023-10-19 by github-actions[bot]