prusa3d/Prusa-Firmware #2545: [FEATURE REQUEST]"Processing..." screen

espr14, 2020-03-21:
i3 MK3S v3.9.0 RC1

Once you click to print a file from the SD card there is around 1 second gap. That feels like the FW/printer is stuck.

Can you add screen with some text ("Processing...") which will show immediately after this click? This will show that the click was received and file is being read so the user can calmly wait.

You could add this after all clicks or just before procedures that start with a little delay. This can cause some kind of flickering or blinking of the screen but it is far better that no change at all.

leptun, 2020-04-01:
What do you think of this implementation? https://photos.app.goo.gl/8BciSwxDs8iHfGcV9
It shows a progress bar during the whole check_file() process.
You may notice that the checking ends at ~50%. This is accurate as that's where the `M84` gcode is located in the searched area.
Also, at the end of the video you get a glimpse of another issue. It is fixed by https://github.com/prusa3d/Prusa-Firmware/pull/2465

PR with the presented implementation can be found here:

espr14, 2020-04-02:
That's great!

PR #2573 (2020-11-13): ✨Progress bar for check_file()
https://photos.app.goo.gl/8BciSwxDs8iHfGcV9

Fixes #2545
closed 2020-11-13 by DRracer