prusa3d/Prusa-Firmware #2546: [BUG] BELT TEST ERROR:  Selftest Error Please Check: X Axis Length

Photogad, 2020-03-21:
When using the "Belt Test" option in the calibration menu, the printer quickly stops the test and says "Selftest Error: Please Check: Axis, Axis X" 

**Please note, all my XYZ calibrations work fine and I'm able to print things A-OK. Running the "Selftest" options passes with no errors**. I am only getting this error when I try to run the Belt Test.

**Printer type** - MK3
**Printer firmware version**-  3.9.0 RC1

**Describe the bug**
When using the "Belt Test" option in the calibration menu, the printer quickly stops the test and says "Selftest Error: Please Check: Axis, Axis X" 

Please note, all my XYZ calibrations work fine and I'm able to print things A-OK. Running the "Selftest" options passes with no errors. I am only getting this error when I try to run the Belt Test.
  
**To Reproduce**
1. Have my printer
2. Run the belt test in calibration menu
  
**Expected behavior**
It gives me the belt tension numbers without failing.

Photogad, 2020-03-21:
Also I have used this Belt Test option before in firmware 3.9.0 BETA with no problems, I only get this error after upgrade to RC1.

ochm, 2020-03-21:
I saw the same problem but only when on X was some obstacles at the right side, and the length of axis was not so long as expected during belt tension measurement.
IMHO the belt test goes well (on both my printers).

Photogad, 2020-03-21:
> 
> 
> I saw the same problem but only when on X was some obstacles at the right side, and the length of axis was not so long as expected during belt tension measurement.
> IMHO the belt test goes well (on both my printers).

I don't believe I have any obstacles, I'm not sure what you mean. The selftest goes fine.

For the Belt Test, it doesn't even try to start it. I start belt test button, it says "checking x" for a few seconds, then the print head moves up about 20mm and gives me the error. It doesn't even try to move the X axis at all to test it.

leptun, 2020-03-22:
Ah, yes. This bug. We know about it. It is because of the stealth mode of the motors. Try switching to normal mode and try again.
A fix is planned for 3.9.0 final. #2492

Photogad, 2020-03-22:
> Ah, yes. This bug. We know about it. It is because of the stealth mode of the motors. Try switching to normal mode and try again.
> A fix is planned for 3.9.0 final. #2492

Thats it! I do use stealth mode.

Thanks.. Closing issue.

closed 2020-03-22 by Photogad