prusa3d/Prusa-Firmware #2548: [FEATURE REQUEST] MMU2 random color change

flinke-flasche, 2020-03-22:
is it possible to integrate a random color change for the MMU2 like the Palette 2?

Maybe it's possible to insert some values like 
- how often you like to change
- which of these extruder you like to use -> 1, 2, 3, 4, 5 (checkboxes for each extruder)
- min. Number of lines each color
- max. Number of lines each color
- purge on/off

vintagepc, 2020-03-23:
This would need to be built into the slicer (PrusaSlicer) as it would need to account for the wipe tower etc - I'd suggest you move your feature request there.

The printer firmware alone cannot do that. (sorry)

flinke-flasche, 2020-03-23:
Oh,  yes you're right. i thought i posted it on Prusa Slicer.

closed 2020-03-23 by flinke-flasche