prusa3d/Prusa-Firmware #2550: [BUG]Display shows 0.15 mm instead of 0.12

espr14, 2020-03-23:
i3 MK3S FW 3.9.0 RC1

The following code starts printing at 0.12 mm Z but display shows 0.15 instead.

```
; generated by PrusaSlicer 2.2.0-rc5+win64 on 2020-03-23 at 12:31:20 UTC

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 1.00mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.42mm

M73 P0 R70
M73 Q0 S73
M201 X1000 Y1000 Z1000 E5000 ; sets maximum accelerations, mm/sec^2
M203 X200 Y200 Z12 E120 ; sets maximum feedrates, mm/sec
M204 P1250 R1250 T1250 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X8.00 Y8.00 Z0.40 E1.50 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M107
M862.3 P "MK3S" ; printer model check
M862.1 P0.4 ; nozzle diameter check
M115 U3.7.2 ; tell printer latest fw version
G90 ; use absolute coordinates
M83 ; extruder relative mode

M104 S215 ; set extruder temp
M140 S60 ; set bed temp

G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Y-3.0 Z1 F1000.0 ; go outside print area

M190 S60 ; wait for bed temp
M109 S215 ; wait for extruder temp

G92 E0.0
G1 Z0.12
G1 X60.0 E9.0 F1000.0 ; intro line
M73 Q0 S73
M73 P0 R70
G1 X100.0 E12.5 F1000.0 ; intro line
G92 E0.0
M221 S95
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
M900 K30 ; Filament gcode
;BEFORE_LAYER_CHANGE
G92 E0.0
;0.12
G1 Z0.120 F12000.000
;AFTER_LAYER_CHANGE
;0.12
```

leptun, 2020-03-23:
The firmware has a minimum Z position of 0.15mm This is so that low layer heights on the first layer will not damage the bed.

espr14, 2020-03-23:
So this results in slight under extrusion or the FW compensates higher Z by higher extrusion?

Should this be considered in the PrusaSlicer? Something like "Minimal Z" in Printer Settings -> General?

vintagepc, 2020-03-23:
There's no compensation. It will just print as normal but refuse to move below 0.15.

closed 2020-04-17 by espr14