prusa3d/Prusa-Firmware #2552: [ENHANCEMENT] Skip preheat for MMU eject when filament not loaded

NielschB, 2020-03-24:
Enter what type of printer or upgrade the enhancement applies to.
MK3S + MMUxS

**Is your enhancement related to a problem? Please describe.**
When selecting eject filament from the menu, the firmware forces the nozzle to be heated end executes the ramming sequence. This is not required when there is no filament loaded in the extruder which saves significant amount of time.
  
**Describe the solution you'd like**
Check for filament load status, if not loaded, skip the preheating and ramming part.

Prusa-Support, 2022-07-23:
Thank you for reaching us.

If I understand correctly, in such a scenario the filament could be simply pulled from behind of the Multi Material, as if you were about to take the spool off. If so, there would be no need at all for the printer's intervention: you can remove the filament entirely by hand.

Please use "eject filament" function only if the filament is actually loaded into the nozzle after a filament scrap is used up.
I'm sorry, off the top of my head I can't seem to find a different case of use - unless a very short piece of filament was loaded by mistake, so short that is not even possible to start a print with it.

Please clarify or close this issue if you are satisfied with the answer. Otherwise, the issue could be closed as stale sometime soon.

Michele Moramarco
Prusa Research

gudnimg, 2023-07-03:
This will be fixed once 3.13.0 is released.

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

closed 2023-07-31 by gudnimg