prusa3d/Prusa-Firmware #2553: [BUG] Extruder x-axis colision error - 3.9.0-RC1

mzsergiu, 2020-03-25:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2
**Printer firmware version**-  3.9.0-RC1

**Describe the bug**
During a print, at some point extruder on x-axis hit far right (collision error) and will not move from there, he tries to move far more right).

He informed me that it detected a crash and if I want to resume the printing, but same behavior, he tries to go more right on x-axis.
I did a manual move of the extruder to left. Rebooted the printer, and did a shelf test. 
Everything works ok now.
  
**To Reproduce**
  I cannot reproduce, it printed already without reboot of printer 5 times same gcode, I got error on 6th attempt to print.
  
**Expected behavior**
  To finish the part :)
  
**G-code**
[top_0.3mm_PLA_MK3S_2h9m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/4380052/top_0.3mm_PLA_MK3S_2h9m.gcode.zip)

vintagepc, 2020-03-26:
This sounds more like an issue with your bearings or hardware than a firmware issue. 

Does your X axis move freely if you try to (gently) move it by hand with the printer powered off? There should not be any roughness or binding. Be sure to lubricate your rails and bearings as noted in the periodic maintenance guide, bearings will run dry over time.

mzsergiu, 2020-03-26:
> This sounds more like an issue with your bearings or hardware than a firmware issue.
>
> Does your X axis move freely if you try to (gently) move it by hand with the printer powered off? There should [...]

I can move by hand by rotating slowly the motor, as i know that he produces power. No hardware issue. Since then printer is working from 24h now, i stop it like 10 minutes to cool down a bit, did not encounter any more collision till now.

As I said, he tried to move more into the right, even if he was maximum right. Like he lost his position on the axis on software level.

vintagepc, 2020-03-26:
Yes, skipped motor steps will cause that, and those are the result of motion system issues, especially if the issue is not reproducible with the same g-code. Do you run with crash detection turned on in Normal mode? 

Try that - if your printer starts detecting crashes and re-homing then you have issues with your motion system that need to be addressed. 

Please try moving the extruder itself by hand, not the motor. Turning the motor applies far greater torque and you don't get a good feel for how smoothly the assembly is sliding. 

Additionally, check your smooth rods for buildup of debris and dust, be sure they are clean. Then, inspect them for scoring/scratches. Those are a sure sign you have a failed bearing. 

Remember, the printer has no absolute position feedback loop. The only time it knows for sure where it is is during a home since that takes it to 0,0,0. Any other time, it is just assuming that things have moved as directed (excepting crash detection handling), and the position it thinks it is at is the end result of all the moves it has made since it last homed. If one of those moves does not go as expected...

mzsergiu, 2020-03-27:
I checked both of the extruder axis, no scratches. is brand new the printer, but anyhow no scratch.
I pushed gently the extruder with finger from far left to right, it moves, but i was not satisfied, is not smooth enough, so on left motor i re-calibrated the shaft when belt is, pulled it a bit on exterior but it does not touches the walls, it`s smoother now. Did same for heating plate motor.
The belt is pretty tight, if i press with fingers is bends but not easy, it`s not flimsy.

I run crash detection on and printer is in normal mode (not stealth).

There are any software situations that could lead to this? I understand that he start at 0,0,0 and there is not way for him to know where he is except when he hits the end with collision detection.

vintagepc, 2020-03-27:
I am not aware of any situations in SW that would cause this mid print as you describe; I suspect we'd be getting a lot more reports of the same issue if this was truly a problem in the code - and it would be more reproducible.

Good to hear you did find some potential causes and corrected them.

Your belts might be a bit too tight based on your description, that can also cause issues. 
If you have access to a force gauge or trigger pull gauge, park your extruder at the extreme left, and the bed at the rear of the printer. If you apply 200g of force to the middle of the lower belt on X and Y, you should get somewhere between 5-6mm of deflection.
That should get you tight belts that are not too tight as to exceed the side-loading specifications of the motors.
Adjusting the pulleys may also have helped, it's not uncommon for new builders to place them too close to the face of the motor and have it bind there.

mzsergiu, 2020-03-27:
I will close the issue, I have enough information for now. Thanks!

closed 2020-03-27 by mzsergiu

vintagepc, 2020-03-27:
Glad I could help!