prusa3d/Prusa-Firmware #2555: [BUG] uncontrolled movements of the stepper motors

ixtrader, 2020-03-25:
When selecting "Print from SD card" and navigate to deeper subdirectory some of the stepper motors are moving in arbitariy directions. The only way to stop this ist to restart the system.

Software version prusa3d_fw_3_8_1_MK3.
Printer MK3 no upgrade
I can reprode the problem with the following steps using the 
[https://forum.prusaprinters.org/wp-content/uploads/wpforo/default_attachments/1585079980-prusaMK3SelectProblem.zip](directory structure)

1. Powering mk3 up and inserting SD Card.
2. navigate trough directory of SD Card. 
3. When reaching Directoy
      \1\2\3\4\5\MasterSpool_V4_002_Export3_PARTA_RTP-pet.gcode
   the steper motors are starting to move uncontrolled.

I investigated the problem a little bit deeper. 

    Reformatted SD Card.  -> problem exists
    rewirtten Firmware 3.8.1 -> problem exists
    downgrade to Firmware 3.7.2 -> problem resolved

by downgrading to Firmware 3.7.2 the problem was resolved.

There might be a bug in your memory management.

leptun, 2020-03-26:
Does the problem go away if you use 3.9-RC1?

vintagepc, 2020-03-26:
For reference I took a look at this in the original forum thread and I could not repro the issue with the current RC:
https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/uncontrolled-movements-of-the-stepper-motors/#post-201736

ixtrader, 2020-03-27:
I tried the current 3.9.0 RC1. Within this version the printer can select the problematic Directory.
But you should investigate why the problem exists in Version 3.8,1. The problem might be serious and can affect a complet other funtion in the new RC1.

vintagepc, 2020-03-27:
It just occurred to me that at some point @leptun showed me a video with something similar going on, so if it's no longer occurring then I am reasonably confident that the underlying issue was found and fixed.

TheDuggem, 2020-03-28:
It sounds similar to #2237

leptun, 2020-03-28:
It sounds really similar and that’s the issue. I’ve already fixed it before, but somehow it came back because of something else.
I have a suspicion of what might go wrong, but I’m not sure.

closed 2023-10-24 by github-actions[bot]