prusa3d/Prusa-Firmware #2559: [BUG]<Won't Resume After A Layer Pause>

Jflesch61390, 2020-03-27:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S
**Printer firmware version**- 3.8.1

**Describe the bug**
I put in a pause to insert a magnet. The print paused properly and I inserted the magnet but when I pressed return to print it touched off and wouldn't resume printing. It repeated the pause.
  
**To Reproduce**
  Run the print and after the layer pause it wont resume.
  
**G-code**
[Embed.Mag.Bit.Driver.JFv3_0.15mm_PLA_MK3S_2h32m.gcode (1).zip](https://github.com/prusa3d/Prusa-Firmware/files/4395695/Embed.Mag.Bit.Driver.JFv3_0.15mm_PLA_MK3S_2h32m.gcode.1.zip)

leptun, 2020-03-27:
The issue is fixed in the latest 3.9 RC.

tuxmobil, 2020-04-02:
I just experienced something similar. MK2.5S with 3.8.1. I used the newly added feature in PrusaSlicer to add a pause. I did use Octoprint via USB. Print Paused and after I resumed print in Octoprint, the print actually resumed, but the extruder didn't move at all. So it printed air instead.

I checked the gears visually. No movement at all (so unlikely clogging was not the issue).

Is this also covered by the fix in 3.9RC?

leptun, 2020-04-02:
It should be this as well.

x-c-dog-x, 2020-04-03:
I'm having this happening on 3.90 RC1 on a stock MK3S printing from Slic3r 2.20 gcode.

wavexx, 2020-04-16:
@x-c-dog-x can you share some gcode where this happens?

Eenie75, 2020-05-10:
Hey guys, I had the same issue with mine, I had a colour change (which worked fine) and then a pause so I could drop in an insert.

When it got to the pause, it wouldn't unpause afterwards, it just moved back over (and turned down the hot end temp to 0). I tried resuming like twice (on the printer) and it didn't work.  I was running it through Octoprint, so I tried clicking resume on Octoprint and it restarted, but I hadn't seen it has lowered the Nozzle temp (and by then it was too cool) so that print was lost.

I just retried it again using the SD card thinking maybe it was Octoprint for some reason, and the same thing happened.  This time I managed to get it to work by re-preheating each time the pause reset the temp, and after like 4-5 resumes the print finally continued.

[Voronoi Heart - Empty_0.2mm_PETG_MK3S_6h48m.zip](https://github.com/prusa3d/Prusa-Firmware/files/4604988/Voronoi.Heart.-.Empty_0.2mm_PETG_MK3S_6h48m.zip)

addame, 2020-05-14:
Hi, instead of just pausing to insert magnet, I tried to pause to change color and it worked well! Another advantage of pausing that way: it beeps.

Eenie75, 2020-05-19:
> Hi, instead of just pausing to insert magnet, I tried to pause to change color and it worked well! It paused, told me to change the color, I simulated changing the color, than I resumed it.
> 
> Another advantage of pausing that way: it beeps.

Only downside of doing it this way is that it ejects and you have to reload the filament.
Not a big deal for one, but if you have a few pauses it makes it a pain :)

Eenie75, 2020-05-19:
Just downloaded prusa3d_fw_3_9_0_MK3S and it seems to fix the problem perfectly :)

Jflesch61390, 2020-11-25:
> Hi @Jflesch61390 can you please confirm or reject that this issue is fixed in 3.9.0? Thank you

It is fixed. 

Josh Flesch

JohnnyDeer, 2020-11-25:
@Jflesch61390 then i close this issue, thank you :)

closed 2020-11-25 by JohnnyDeer