prusa3d/Prusa-Firmware #2566: [BUG]SUPER slow speed with new firmware

forestqc69, 2020-03-30:
MK3 - 3.9.0-RC1-3272

I've been printing the same piece for 1 week, after upgrading to the current RC firmware, print speed is SUUUPER slow, the time remaining countdown didn't move for 5 minutes or more.  Non-print Movements are as fast as usual I would say.. And yes, print speed is 100%.

[3xtubeV2_Mod1_0.6n_0.2mm_PLA_MK3S_5h37m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/4403523/3xtubeV2_Mod1_0.6n_0.2mm_PLA_MK3S_5h37m.gcode.zip)

leptun, 2020-03-30:
How does it manifest? Really low acceleration. I can't test the gcode at the moment sine all my printers are printing face shields.

forestqc69, 2020-03-30:
I flashed back to 3.8.1 and regular speed resumed, I'd say it like 10-15 time slower on print movements, non-print movement seemed at regular speed.  It's so slow I can't even see accelerations, it just all printing movement that are so slow.. I did put the speed to 120% and it wasn't even close to going at the regular speed..

DRracer, 2020-03-31:
@forestqc69 what nozzle are you using?

DRracer, 2020-03-31:
@forestqc69 
original gcode - really slow
E jerk set to 4.5 in gcode - no change
E jerk 10 in gcode - a little less slow
change from K15 to K30 - normal speed
change from K15 to K0.03 - normal speed
It looks like K15 is being converted to some bad value, we'll look at it, thanks for reporting ;)

forestqc69, 2020-03-31:
0.6mm

wavexx, 2020-04-01:
Related  #2570

sn4k3, 2020-04-05:
yeah that also happen to me with my ABS profile when i had K=15, had to return to 18 to be able to print. Bad conversions...

commit 13b0e27c (2020-04-15): Do not overflow during LA acceleration limiting

Perform the check one step earlier, avoiding 32bit overflow for very low
compression factors.

Fixes #2566 (although for K15 to have effect the conversion probably
needs to be adjusted on the low end)
closed 2020-04-15 by DRracer