prusa3d/Prusa-Firmware #2570: [BUG]Division by zero with "M900 K=15"

smurfix, 2020-04-01:
**Printer type** - MK3S
**Printer firmware version**-  current MK3 master

**Describe the bug**

This code results in `new_K` =0 when `k<=15`. This later results in a division-by-zero error and a printer reboot.

```
  // Approximate a LA10 value to a LA15 equivalent.
static float la10c_convert(float k)
{
    float new_K = k * 0.004 - 0.06;
    return (new_K < 0? 0: new_K);
}
```

**To Reproduce**
Print with M900 K=16`. Works. Now print with `K=15`. Crash.

wavexx, 2020-04-01:
Did you find this by chance during a test of because you happened to use a K value below 15 with some material?

wavexx, 2020-04-01:
I'm looking at this along with #2566. Do you have a short gcode that makes the print crash/reboot? I'd love to add this as a test case.

smurfix, 2020-04-02:
I am printing with a 0.8mm nozzle and noticed splodged bottom infill at the edges even when the overlap is set to zero. The gcode that comes with PrusaSlicer reduces K to 18 when the nozzle is set to 0.6 so I tried with 15, which then crashed before emitting a single blob of filament. Editing the gcode to K=16 rescued the print but still has a bit too much infill.

Thus I assume that basically any print will die when you set K=15.

wavexx, 2020-04-02:
> Thus I assume that basically any print will die when you set K=15.

Not always. Note that the entire range down to K=0 is actually a legitimate value for LA. There is a problem later on that can cause either a zero division or (in case of the other issue) excessive loss of precision, but it also depends on the segment at hand.

I can reproduce nicely the issue in #2566 but couldn't get a crash right away

smurfix, 2020-04-02:
OK. I'll try to recreate the offending file.

vintagepc, 2020-04-06:
Just to clarify here... you are setting `M900 K15` right? Not `M900 K**=**15`? I ask because of the quotes in the title. `K=15` is not valid gcode.

smurfix, 2020-04-06:
Yes, M900 K15, sorry.

UltiCa, 2020-04-11:
I have the same problem with M900 K15 on my prusa. I do not know, if there is a division by zero problem, but if I set M900 K15, then next command cashes the Prusa (MK3S with MMU2S and 3.8.1). He does nothing, the display is frozen 1-2-3 seconds later the Prusa reboots.

wavexx, 2020-04-11:
Are you using the latest beta (3.9.0 RC1) or the latest release (3.8.1)?

smurfix, 2020-04-19:
@wavexx  Umm, neither. I wrote "current master". At that time that was commit e481013.

wavexx, 2020-04-19:
The latest fixes for LA got merged on friday, this should be now fixed in master

closed 2020-09-21 by DRracer