prusa3d/Prusa-Firmware #2578: [BUG] Print fan error on color change

kripobs, 2020-04-05:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

MK3
downloaded the mk3 branch yesterday (04.04.2020)
firmware says:
FW_VERSION "3.9.0-RC1"
FW_COMMIT_NR   3272

**Describe the bug**
When ever the printer Stops (filament problem or color change where my two cases) and starts again, after a few seconds of printing i get a "Print fan error" and cant resume the print. The fan is just fine. i even can (when the printer wont resume) set in the tune settings the fan on any value and it spins just normal. when i try to resume it wont print again and stops the fan.
  
**To Reproduce**
maybe cut the filament or print something with color change?

leptun, 2020-04-05:
Can you please provide a video with the scenario?
What print fan speed were you using when you generated the error?
I've been printing PLA and PETG parts on my MK3 printers with the latest changes and I didn't have any issues. Fan speeds used are 100% (PLA) and 30% (PETG)

kripobs, 2020-04-05:
wow, that was fast :)

kripobs, 2020-04-05:
but i can try to print something with color change at a fixed fan speed. just give me a few hours.

kripobs, 2020-04-05:
interestingly i had the same error between objects. i try the same print now with a fixed fan speed at 70%

leptun, 2020-04-05:
I also had the problem when I upgraded my MK2S to MK2.5S. I just have up and installed a sock. I was printing at extreme speeds anyway, so it would only help.
I'm really surprised that no hardware update was done in the meantime to address this issue. It's quite serious tbh.
Is it MK3 or MK3S in your case? For me, only the R4/5 revision of the extruder had the issues. R3 was working just fine (and still is).

kripobs, 2020-04-05:
ok, so it was a dolly type at the start wich i updated with the original mk2.5 kit. but i ultimately bought a einsy board so at the end its mk3ish

leptun, 2020-04-05:
Same in my case for one of the printers (except it was a genuine MK2S) and for another one I tried to hack myself TMC drivers on the old MiniRambo board so that I wouldn't have to buy a new Einsy. It was kinda a success, but I'm still not fully satisfied with it. The project is plagued with random issues...

kripobs, 2020-04-05:
so one short test with a print of two parts and fixed fan speed and it didnt stop between the parts. so i tried a two color print.
i was just standing by the printer when it failed with the same error. i am not sure if it was the last layer of the first color. could have been the second last layer. but it failed: the fan was spinning while the print failed!
very strange. i will try one last print with four parts in series and fixed fan speed and will after that most probably revert to an older firmware.

kripobs, 2020-04-05:
ok, i get the error a bit random now. i will try an other version. the last firmware i used was 3.8.1 but there i had totally different problems with the first layer.... the last error free firmware version was 3.5 :/

kripobs, 2020-04-06:
I have now installed this version and the errors seem to have disappeared. "3.9.0-RC1" 3272

vintagepc, 2020-04-06:
I think the fan checks may have been a relatively recent addition. Since it's arbitrary and intermittent it sounds like there might be an intermittent break in your tach sense wire of the print fan

kripobs, 2020-04-06:
@vintagepc Admittedly, I would not rule out a cable break. I've been printing all day with version 3.8.1 and have no fan errors. But will try the new version again tomorrow. I have recalibrated my printer and it is generally running much more consistently. So have high hopes for 3.9.0 :D

jltx1, 2020-05-09:
The tach sense appears broken on 3.9.0 RC3

I have been printing fine.  I installed 3.9.0 RC3 (coming from 3.8.1) as a test.  The first print failed quickly with a print fan error.

Setting the fan to 127 in "temperature/fan" menu, I see in "support/extruder" menu the RPM is > 4800 and then setting fan to 255 I see RPM is bouncing around 0 to 200.

jltx1, 2020-05-09:
Just noticed that the nozzle fan is also misreading at 8700 RPM, although it is stable and does not trigger an error.  Noctua fan is 4500 RPM.  

So both tach sense are broken.  Workaround: I can print by turning off fan check.

3d-gussner, 2020-05-11:
@jltx1 Did you use the https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.9.0-RC3 or did you compile the firmware yourself?

jltx1, 2020-05-22:
I compiled the 3.9.0-RC3 tag.  Why, is there a known issue there?  I will try to final release now

3d-gussner, 2020-05-22:
Do you have a stock prusa printer?
Any Mods, other Fans,...?

jltx1, 2020-05-22:
just tried 3.9.0.  I reloaded the board manager in Arduino as it looked stale / wrong.  Now it works.  So I think somehow the board definition was whacked on my end.  Sorry for the false alarm.

3d-gussner, 2020-07-13:
@kripobs I think we fixed this issue, please confirm and close this issue.
[stale][fixed]

kripobs, 2020-07-13:
I havent had a chance to test it with a new fan (its seems both was true for me, software problem and pinched cable) but sure, i close it. Thank you all for your work and support.

closed 2020-07-13 by kripobs