prusa3d/Prusa-Firmware #2579: [QUESTION] Code weirdness in G80

leptun, 2020-04-05:
```
		if (degHotend(active_extruder) > EXTRUDE_MINTEMP && temp_cal_active == true && calibration_status_pinda() == true && target_temperature_bed >= 50) {
			current_position[E_AXIS] += default_retraction;
			plan_buffer_line_curposXYZE(400, active_extruder);
		}
```
This snippet of code at the end of G80 makes no sense. Why only unretract when Temperature calibration is used?

Panayiotis-git, 2020-04-05:
At the beginning of the G80 command there is the following code:
```
#ifndef PINDA_THERMISTOR
		if (run == false && temp_cal_active == true && calibration_status_pinda() == true && target_temperature_bed >= 50)
		{
			temp_compensation_start();
			run = true;
			repeatcommand_front(); // repeat G80 with all its parameters
			enquecommand_front_P((PSTR("G28 W0")));
			break;
		}
        run = false;
#endif //PINDA_THERMISTOR
```
The `temp_compensation_start()` function, does a retraction:
```
	if (degHotend(active_extruder) > EXTRUDE_MINTEMP) {
		current_position[E_AXIS] -= default_retraction;
	}
	plan_buffer_line_curposXYZE(400, active_extruder);
```

leptun, 2020-04-06:
Thanks, that makes a lot more sense.
Now that I think about it, why don’t we always retract during MBL? It just seems logical doing so.

vintagepc, 2020-04-06:
On the topic of hidden gems, I found this lurking in lcd.cpp:

```
#ifdef VT100
 uint8_t lcd_escape[8];
#endif
```
What is a VT100,you ask? 🤣 
https://en.wikipedia.org/wiki/VT100

leptun, 2020-04-06:
@vintagepc it’s not hidden code, it is my code :).
The vt100 codes are escape sequence of characters that trigger a certain terminal command. In our case, we had home and set home that were useful for the lcd.
These were quite wasteful, so I changed the code to not use them anymore and they were no longer needed. I commented out the code so that I could save close to 1KB of flash

vintagepc, 2020-04-06:
Interesting - either way it gave me a good laugh!

closed 2023-08-17 by DRracer