prusa3d/Prusa-Firmware #2580: [BUG] Mesh Bed leveling Test uses hardcoded values

Khazrak, 2020-04-05:
**Printer type** - MK3S
**Printer firmware version**- 3.9.0 RC1 (and all earlier

**Describe the bug**
Mesh Bed level Test uses hardcoded values instead of variables like MESH_BED_X. This will let it travel outside of X_MAX_POS. This is only a problem if you have changed the frame or bed (I have Bear Upgrade and Hemera upgrade)
  
**To Reproduce**
Need a frame that allows sensor to go outside of bed and then run Mesh bed level Test.
  
**Expected behavior**
The class "mesh_bed_calibration.h", line 9-12 should use variables from Firmware Variants instead.
ex: #define BED_Xn (206.f - BED_ZERO_REF_X) //205
Should use "MESH_MAX_X" and some calculations.

ManuGithubSteam, 2020-04-19:
Hi

Printer type - MK3S
Printer firmware version- 3.9.0 RC1 (and all earlier

Im hit by this as well. I did not do the bear upgrade but i did the steel springs unter the bed. After that i have a hard time with the pinda finding the bed. As the springs have a different height as the normal distancers. I very often get: Sensor did not trigger. Debris on nozzle.... etc. Of course the nozzle is clean but due to the different height of the bed it does not find the bed. So it should use the values it gets from the xyz calibatrion on wich Z coordinate the bed is and not hardcoded ones.
Thanks!

JohnE334, 2020-07-10:
Printer type - MK3S
Printer firmware version- 3.9.0 RC1

Describe the bug
Changing  MESH_MAX_X  variable doesn't affect mesh bed leveling. My bed is shorter in the x direction than Prusa i3 MK3 bed, but I am unable change this variable.
To Reproduce
Change MESH_MAX_X variable and watch mesh bed leveling to see if it changes in the x direction.

Expected behavior
Changes to "MESH_MAX_X" variable should change the mesh bed leveling in the x direction.

closed 2023-10-24 by github-actions[bot]