prusa3d/Prusa-Firmware #2581: [BUG] MMU2S Continious Reset Loop

Cx01N, 2020-04-06:
**Printer type** - MK3S
**Printer firmware version**-  3.9.0 RC1

**MMU Upgrade** -MM2S
**MMU upgrade firmware version 1.06

**Describe the bug**
The MMU continuously loops and resets whenever I print.
  
**To Reproduce**
Start a print or first layer calibration and the MMU2S will loop with restarts. The easiest way to fix it is to press the button on the side of the mmu. This only sometimes works, other times the mmu will get started back up without motor control without a way to restore it without a complete power cycle.

Cx01N, 2020-04-06:
Here is a sample of the gcode from my most recent print.
[3DBenchy_0.15mm_PLA_MK3SMMU2S_12h11m.zip](https://github.com/prusa3d/Prusa-Firmware/files/4435621/3DBenchy_0.15mm_PLA_MK3SMMU2S_12h11m.zip)

vintagepc, 2020-04-06:
What lights are blinking on the MMU when it resets? 
If it's all 5 red and green at the same time your tension springs may be too tight or the power connection is not secure.

Cx01N, 2020-04-06:
@vintagepc It is all 5 lights but they are red only. I tried adjusting it a few different times and didn't have any luck.

vintagepc, 2020-04-06:
Red only suggests the FINDA is detecting while trying to move the selector - Is that what you're saying you tried to adjust?
Does the FINDA value in the support menu agree with what it should be reporting?

Cx01N, 2020-04-06:
Sorry. I was very not descriptive about that.
I adjusted the springs, but that did not fix the problem.

As for the FINDA value, it shows a 0 when the 5 red lights were up. I tried cleaning out the MMU of any debris, but the issue is still persisting.

vintagepc, 2020-04-06:
Hmm.. What about the light on the FINDA? It's there but dim. TBH I think there's something specific with your setup going on; I have an MMU myself and have used it for filament changes and such without issue on both 3.8.1 and 3.9-RC. So trying to narrow it down to what might be going on with yours.

martysweet, 2020-04-07:
I also had this issue yesterday with 1.0.6-372. All 5 red and green lights started flashing after an unload attempt and the MMU2 unit kept almost restarting. There was a faint orange glowing LED somewhere on the board also.

Like most issues I've been having since 1.0.6, power cycling the printer fixed the issue without any further changes. Hopefully this helps narrow down the problem!

Cx01N, 2020-04-08:
It looks like when I go back to 3.8.1 the issue goes away. Not sure what's happening in 3.9-RC1.

closed 2023-10-24 by github-actions[bot]