prusa3d/Prusa-Firmware #2582: [Question] Possible undefined timer behaviour

vintagepc, 2020-04-06:
Found while investigating a bootloop with simulated timers. I'm not sure whether this is of concern since it clearly works in practice, but seems like it has the potential for odd problems:

In setup() in Marlin_main.cpp:
line 1010 of ultralcd_init() calls backlight_init() which requests _delay(10) after setting LCD_BL_PIN.

However, timer2 (whose TOIE is used for the delay) is not actually initialized until tp_init() (Line 1241) of the same function.

leptun, 2020-04-09:
Maybe the simulated timer has a different initial state than the actual hardware.

vintagepc, 2020-04-09:
Probably - though I searched and only found a blanket statement that they are disabled until set up by default. 

Thoughts on where I might be able to find the initial state?

closed 2020-04-12 by vintagepc