prusa3d/Prusa-Firmware #2583: [BUG] v3.9.0-RC1 is ~30% slower than slicer estimate and v3.8.1

titanous, 2020-04-07:
**Printer type** - MK3S
**Printer firmware version**-  3.9.0-RC1

**MMU Upgrade** - MMU2S
**MMU upgrade firmware version** - 1.0.6

**Describe the bug**

I'm printing face shield stacks. The print time is around 30% slower on v3.9.0-RC1 than the PrusaSlicer v2.2.0 estimate, which is accurate for firmware v3.8.1.
  
**To Reproduce**

1. Download a [RC3 stack 3mf](https://www.prusaprinters.org/prints/27267/files).
1. Change the filament custom g-code to `M900 K0.2`
1. Change the machine_max_jerk_e to 4.5
1. Save g-code to SD card and print.
1. Note that the print time is slower than the estimate.
  
**Expected behavior**

The print completes in the estimated time.
  
**G-code**

[8xfast_covid19_rc31_8pcs_complete_0.25mm_PET__10h37m.zip](https://github.com/prusa3d/Prusa-Firmware/files/4441920/8xfast_covid19_rc31_8pcs_complete_0.25mm_PET__10h37m.zip)

wavexx, 2020-04-09:
In the new firmware we're using a different version of linear advance which changes the printing time of some moves.

We tried to match the print time between these versions. However I'm not sure though if PrusaSlicer can handle the new meaning of the K advance factor for correct time estimation (that is, the estimation itself might be off)

titanous, 2020-04-09:
Yeah, I think there could be two issues here. The same gcode is taking quite a bit longer with LA1.5, and PrusaSlicer is not be estimating it correctly.

DRracer, 2020-04-16:
@titanous for extremely fast prints you may bump the E-jerk way beyond the usual 4.5. Actually, the 3.9 FW can get faster than the 3.8.1.

wavexx, 2020-04-16:
@DRracer I'd also love to know how far can you push e-jerk, don't keep us hanging ;)

DRracer, 2020-04-17:
@titanous @wavexx 
So far I only have one set of tests based on one original G-code with LA10 settings. This G-code was tuned for fastest print time already.
- With FW 3.8.1 the total print time was **3h 43m** and there were several skipped steps of the E-motor during the print (LA10 doesn't obey E-jerk at all).
- With FW 3.9 and **E-jerk=8** I was able to run the same G-code with total print time **3h 40m** with no E-motor skips and yet visibly smoother motion of the whole machine.

We selected E-jerk=4.5 from our set of comparison tests on existing G-codes which ran on our testing farm for the past half year. Fine tuning of the LA15 is still to be performed in greater range of applications, I believe, the machine could go even faster.

DRracer, 2020-09-22:
@titanous is this issue still present in FW 3.9.1 final? If not, can we close the issue?

JohnnyDeer, 2020-11-25:
Closing this issue as stale.

closed 2020-11-25 by JohnnyDeer