prusa3d/Prusa-Firmware #2590: [ENHANCEMENT]Detect start of X/Y axis by loosing a step

espr14, 2020-04-09:
i3 MK3S FW 3.9.0 RC1

Search for start of X/Y axis is done by crash detection. Therefore, no step has to be lost. However, in case of higher resistance of bearings, this brings bad start position.

If you loose single step you can get these results:
1. Crash detected and step is lost = real start of an axis
2. Crash detected but step is not lost = this is not the start of the axis. The printer could raise false crash detections during printing => inform user.

leptun, 2020-04-09:
Unfortunately, the tmc drivers can’t differentiate a skipped step from a harder surface.
The way these drivers work is that they measure how hard the axis moves. If it is harder than a threshold value set in software, it raises the crash detected flag. The drivers don’t know if a step is skipped or if it’s just a momentary increase in difficulty in moving the axis, only that it surpassed the threshold.
You can see this value in the belt test. The resulted value is the raw value reported by the driver.

espr14, 2020-04-10:
But FW can detect if the curve is exponential or teeth like. Once there are two similar drops, they are step skips.