prusa3d/Prusa-Firmware #2592: [BUG] Strange X & Y offset when printing simple 2 perimeter part

Windscreen2011, 2020-04-10:
MK3S
3.9.0 RC1

While printing a pretty simple part, I've found a strange X/Y offset on seemingly random groupings of layers.  Part was sliced in PrusaSlicer 2.2.0.  .3mf and gcode attached.

When the error happens, a number of layers will be slightly offset toward the center of the part.  After a somewhat random number of these layers the expected printing returns.  You can see the visual negative groove on the part's outside and the positive on the inside.  I thought it maybe was the slicer's fault, but when checking the gocde on a simulator, the tool paths look as expected.

The grooves reproduce the same way, run after run of the same gcode.  I have printed a nest of 22 of these parts, and while some parts print with the same groove defect, others have different grooves, at different Z heights.

One part with the grooves
[image]

You can see the grooves are an offset in the perimeter thru this triangular window on the back side of the part:
[image]

Tool paths look as expected:
[image]

Gcode and .3mf
[5cm_Groove_Troubleshooting.zip](https://github.com/prusa3d/Prusa-Firmware/files/4459626/5cm_Groove_Troubleshooting.zip)

The outer walls of this part should be a smooth, continuous surface.

Windscreen2011, 2020-04-10:
Perhaps another clue from watching how the printer is building the part.  Keep in mind this part is designed to have a wall thickness of only 2 perimeters.  When the part builds as expected, the printer starts on the right, lays down what I would call the inner perimeter, then continues on from left back to right with the outer perimeter.

When the groove happens, the printer starts on the left, lays down the outer perimeter, then continues on from right to left with the inner perimeter.

wavexx, 2020-04-16:
Mmmh, this might actually be due to the extrusion direction (clockwise or counterclockwise). I noticed the 3mf has a .3 layer height, that with a .4 nozzle makes this sort of issues much more visible.

Did you notice if this matches the direction while it's being printed?

Windscreen2011, 2020-04-17:
Yes, it seems related to clockwise vs counterclockwise.  These parts, being 2 perimeters, generally print the inner perimeter first, clockwise, then the outer perimeter, counterclockwise.  When it changes to outer perimeter counterclockwise first, followed by inner perimeter clockwise, the groove occurs.  

At the PrusaSlicer settings for 0.2 mm Speed, this doesn't happen, or at least is not noticeable.  A friend created some settings for 0.25 mm layer height, and the grooves returned.

For what it's worth I know if three local Mk3s doing this on 3.9.0 RC1 with this part.  Two kit builds and one factory assembled.

wavexx, 2020-04-17:
I don't think there's anything wrong with the FW in this case.
The issue is that the order and width of the extrusions matter _a_lot_ when working with a 2 perimeter part, since the first extrusion will inevitably push the second outward.

I do a lot of 2 perimeter parts. The width of the wall is important here. The effect is greatly reduced if you use the same width slicer tells you in the "print settings" tab (look for "reccommended object .."). And also note that this is the width calculated on the horizontal slice of the plane, not the wall width as you would normally assume. This is because the width slicer is telling you is the optimal  space taken by two extrusions side-by-side.

You can see that if you increase the layer height and/or change the extrusion widths in the advanced settings the recommended width changes (this is due to the perimeter overlap).

If you have a wall which is thinner than recommended slicer will still perform two extrusions (up to a point), but one of them will bulge outwards, doing this.

Windscreen2011, 2020-04-18:
Wavexx, you seem to be on the money...it's wall thickness relative to 2x extrusion width.  At 0.3 layer height, the default extrusion width x2 is a fair bit more than this part's wall thickness (recommended wall thickness for 2 lines is 1.14, as best as I can tell in Meshmixer, this object's wall is about 0.85).  So, when the wall is extruded, the 1st line gets placed on location, and the 2nd line gets forced off to the side by the "fat" 1st line.

I verified this by changing all my extrusion widths to 0.45 at layer height 0.3.  The part printed fine, without the groove artifacts.

So, I guess I should close this.  But before I do, do you think this is a worthwhile feature request to implement in PrusaSlicer?  It seems to me like it would be.

Thanks.

wavexx, 2020-04-18:
At least on PrusaSlicer, we have a ton of requests for the "detect thin walls" feature, which essentially should enable variable-width extrusions when necessary, covering cases like this. Pick the one you like the most and join the discussion ;)

I have big hopes for 3+ perimeters being handled better, but for for 1 or 2 perimeter walls even "thin walls" processing cannot do miracles at certain sizes. The reason is that doing thinner extrusions will make the bonding weaker, while doing 2x width extrusions works only for some materials, but still loses a lot of resolution on the corners. Both artifacts are will visible when printing such objects, so if you have the luxury of designing a piece that needs to be thin, considering the extrusion width from the start is important.

wavexx, 2020-04-29:
Unless you have any other doubt, I think you can close this :)

closed 2020-04-29 by Windscreen2011