prusa3d/Prusa-Firmware #2593: [ENHANCEMENT] don't unload filament on MMU2S when first layer calibration finishes

Photogad, 2020-04-10:
**Printer type** - MK3S
**MMU Upgrade** - MMU2S

**Is your enhancement related to a problem? Please describe.**
 When first layer calibration completes it unloads filament. But the calibration is very fast so sometimes have to do it 2 or 3 times to get it perfect. So unloading filament is cumbersome.
  
**Describe the solution you'd like**
  
I understand it unloads to keep the filament tips in shape but maybe it could ask before it unloads, or ask if I want to run first layer calibration again

Mirarkitty, 2020-12-28:
I like this suggestion. Also I would like temperature to stay hot so I can rerun the test.

mcarland, 2021-03-07:
I concur. When the test is done, it should stay hot-n-ready, and ask to run the test again or quit. If you quit, or timeout, then go ahead to cool off/unload filament.

I'm having issues with the first layer adhering, trying to run calibration multiple times, and I'm pulling my hair out with this filament loading. Getting ready to rip out the MMU, it's more trouble that it's worth.

closed 2023-10-24 by github-actions[bot]