prusa3d/Prusa-Firmware #2594: [ENHANCEMENT]Show motor state [on] / [off]

espr14, 2020-04-10:
i3 MK3S FW 3.9.0 RC1

In menu user can switch motors off but it does not show if the motors are on or off. In other words, there's no indication on display whether the switching was performed. Can you add something like [on] / [off]?

vintagepc, 2020-04-12:
When not printing, 99% of the time they are off. They time out after several seconds, and most prints finish with an M84 command that turns them off. 

Maybe I'm misunderstanding but there's not a whole lot of purpose to being able to turn them back on, and the menu doesn't do that anyway, all it does is off, not toggle.

espr14, 2020-04-12:
I didn't meant toggle. Just something that indicates that the motors are still on and after clicking they are off.

closed 2023-10-23 by espr14