prusa3d/Prusa-Firmware #2600: [TranslationFR] Function "Cut Filament"

Forfelet, 2020-04-12:
Hello,
I upgraded my MK3S + MMU2S with the last firmware 3.9.0 RC1.
I'm using french langage package. And, I saw the new function "Cut filament" in the main menu. 
But, in french, this function should be named "Couper filament".
It's a small fix but more user friendly for my compatriots :).

I take advantage of the post to have news on the new selector for the function "Cut filament". 
When will it be available for testing?

Thank you for your job.
Have a nice day. 

Cyril.

Forfelet, 2020-04-13:
Re,
I have the answer for my question #2495. 
Thx.
Cyril

leptun, 2020-04-13:
Hello @Forfelet,
Thank you for submitting this translation. I'll make sure to transmit this request to the correct department.
Firmware translations are usually done on the final release only. Release candidate builds (RC) don't usually get the translated texts for the new features. Unfortunately, because of some unforeseen problems (some LA1.5 issues and the pandemic) the release of 3.9.0 final is delayed.

leptun, 2020-04-14:
@Forfelet 
3 MMU parts have changed:
- https://github.com/prusa3d/Original-Prusa-i3/blob/MMU2/STL/mmu2-blade-holder.stl
- https://github.com/prusa3d/Original-Prusa-i3/blob/MMU2/STL/mmu2-front-PTFE-holder.stl
- https://github.com/prusa3d/Original-Prusa-i3/blob/MMU2/STL/mmu2-selector-finda.stl

Forfelet, 2020-04-14:
Hello @leptun,

Thank you for your answer. 

Best regards,
Cyril

Forfelet, 2020-05-23:
Translation is done in the release version 3.9.0

closed 2020-05-23 by Forfelet