prusa3d/Prusa-Firmware #2601: [BUG?/QUESTION]PINDA temperature calculation adds bed temperature correction factor

vintagepc, 2020-04-13:
In Temperature.cpp, the PINDA and bed use the same function, analog2bedTemp. 

However, this function contains a correction factor for bed temperatures over 40C, that ls also applied to the PINDA temperature. The low end of the curve is quite steep (5 degrees over 10C!!) and PINDA temperatures over 40C are probably not all that uncommon when printing with PETG and ABS. 

I'm guessing this correction is a function of the bed thermistor placement and not the thermistor type itself - or it would have been baked into the temperature tables. 

Seems like this should probably be fixed since it could introduce quite a bit of nonlinearity into things like the PINDA temperature calibration.

chriserx, 2020-04-13:
Could also be why temperature calibration doesn’t work for me in this release, I did modify the Z height to accommodate my supervolcano hot end, otherwise unchanged.

vintagepc, 2020-04-13:
Hmm... I don't think so as it's a positive correction - that is, it reads higher than it actually is by a few degrees, so if anything you'd be able to reach 70 more easily.

chriserx, 2020-04-13:
I haven’t actually looked at the code other than to change the Z height limit, I was just speculating the issues could be correlated.  The bed has no problem reaching the temp, I left the calibration running while I went to sleep, woke up the next morning and bed was still 70 degrees, step 2.  I may mess with it some more later after I finish printing out my other project.  For now
¯\_(ツ)_/¯

leptun, 2020-04-13:
@chriserx The bed calibration "endless" issue is known and it is caused by the calibration algorithm. I'm working on a fix that would also improve the performance of that function.
Here is the PR with the changes. Please keep in mind that it's still experimental. The old values will not be overwritten by the calibration, but you will have to run the new calibration.

closed 2023-10-24 by github-actions[bot]