prusa3d/Prusa-Firmware #2602: [BUG] G28 or G80 moves E when temp is >=180

ochm, 2020-04-13:
I'm not sure if it is a bug or feature, but IMHO mes bed leveling should not to do extrusion at the end.

**Printer type** MK3S
**Printer firmware version**-  3.9.0-RC1-3272
**Describe the bug**
When the nozzle is preheated under 180, the G28, or G80 does mesh bed leveling, 
and the extruder is parked at 0,0. Correct.
But when the nozzle is preheated to or over 180, after G28 or G80, the extruder is parked at 0,0 and additionaly does short move of E. 
  
**To Reproduce**
No E move at the end of mesh bed leveling:
M109 S160
G28 ;or G80

Extra E move at the end of mesh bed leveling:
M109 S180
G28 ;or G80

**Expected behavior**
As i wrote above, i don't know if it is bug or feature, 
but IMHO there is no reason for move E after mesh bed leveling at any temperature of nozzle.

leptun, 2020-04-13:
Isn't the Extruder retracted at the beginning of MBL? Maybe what you are seeing is simply the unretract movement.

Panayiotis-git, 2020-04-13:
IIRC, if the hotend temperature is greater than the `EXTRUDE_MINTEMP`, the firmware retracts the filament at the beginning of the MBL.

ochm, 2020-04-13:
Not, the E moves after MBL and the movement is in forward (extrusion) direction, not retraction.
You can see it at video in a moment...

leptun, 2020-04-13:
@ochm Well, yes. That is what an unretract movement is.
Retraction pulls filament (backwards), unretraction pushes filament (forwards).
I believe this behavior was done so that during MBL the oozing of filament would be minimal.

ochm, 2020-04-13:
http://www.och.cz/tmp/E-moves-after-MBL.mov you can see extraction of some filament after MBL. There is absolutely no reason to do it.

leptun, 2020-04-13:
Hmmm. There is something broken. Only unretracting without retracting first is problematic. I believe that retracting prior to the MBL could be a solution.
The retraction should be done by the firmware automatically. No external gcode required

Panayiotis-git, 2020-04-13:
@leptun, I think we had a discussion on this. That the specific unretraction should be inside an #IF block, similar to the retraction.

Panayiotis-git, 2020-04-13:
At the `Marlin_main.cpp`, there are different conditions on when to do the retract:
```
#ifndef PINDA_THERMISTOR
		if (run == false && temp_cal_active == true && calibration_status_pinda() == true && target_temperature_bed >= 50)
		{
			temp_compensation_start();
...
#endif //PINDA_THERMISTOR
```

than when to do the unretrack:
```
		//unretract (after PINDA preheat retraction)
		if (degHotend(active_extruder) > EXTRUDE_MINTEMP && temp_cal_active == true && calibration_status_pinda() == true && target_temperature_bed >= 50) {
			current_position[E_AXIS] += default_retraction;
			plan_buffer_line_curposXYZE(400, active_extruder);
		}
```

and `PINDA_THERMISTOR` is always defined: Rambo or Einsy

ochm, 2020-04-13:
IMHO no movement of E should be done when doing MBL (prior or after) in firmware.
The good practice is heat the nozzle to temperature not so high (to avoid ozzing), so the the nozzle is heated to 160-180, then goes MBL and after MBL is the nozzle heated to temperature for the first layer. IMHO this is good practice.

closed 2023-08-17 by DRracer