prusa3d/Prusa-Firmware #2603: [BUG]<Selftest failured XYZ Calibration>

sharkfish2526, 2020-04-14:
Firstly, I'm very sorry if my issue is not correct or relate about the BUG topic. And very sorry my story may be longer than other issue due to I have tried many thing for solving my 3D printer. But I have only a few experience for 3D printer and during study, so I need to ask someone in this place help me solve my issue that I think this place has many person who has more an experience for help me. 

My 3D printer is MK3S and upgrade last firmware version 3.9.0 RC1, however, my 3D printer is not orginal from Prusa, it came from clone version from China. By the way, all of components and its dimension similar original from Prusa, so it can operate by prusa firmware. 

**First problem after upgrade firmware:**
After firmware upgaded the 3D printer has advised for selftest the same original prusa procedure by strat from cheking 2 cooling fans, XY axes' belts, Z axis, and XY axes home calibration. When start XY axes home calibration normally the extruder must move to left side(X min) and bed must move to back(Y min) also the both must move by hit and backward(move to min stop 11 mm. and move backward 10 mm. and loop this process from the firmware code) But my printer the extruder move always from min stop each 10 mm. and bed move always from min stop 10 mm. so the result extruder will stop on position X= 160mm. and Y = 160mm. after that the printer do remaining process heat bed and heat extruder then start process XYZ calibration. But when printer start calibration Z axis first, the Z axis just move up only a few millimeter and stop. So after continue for XYZ calibration the LCD has show "Calibration failed please check axes"  that's all first problem for selftest.

**My solving method try to config firmware** 
After I have studied firmware for a month, I have tried edit the firmware for bold text highlight in Marlin_main.ccp as below:

#ifdef TMC2130
bool calibrate_z_auto()
{
	//lcd_display_message_fullscreen_P(_T(MSG_CALIBRATE_Z_AUTO));
	lcd_clear();
	lcd_puts_at_P(0, 1, _T(MSG_CALIBRATE_Z_AUTO));
	bool endstops_enabled = enable_endstops(false);
	int axis_up_dir = -home_dir(Z_AXIS); //+1
	**tmc2130_home_enter(Z_AXIS_MASK); // Change to tmc2130_home_enter(Z_AXIS)**
	current_position[Z_AXIS] = 0;
	plan_set_position(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS]); //Set Z axis position equal zero(0).
	set_destination_to_current();
	destination[Z_AXIS] += (1.1 * max_length(Z_AXIS) * axis_up_dir); //10+(1.1 x 209.85 x 1) = 229.9mm.
	feedrate = homing_feedrate[Z_AXIS];
	plan_buffer_line(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS], destination[E_AXIS], feedrate / 60, active_extruder); //Move Z axis up to max endstop.
	st_synchronize();
	//	current_position[axis] = 0;
	//	plan_set_position(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS]);
	tmc2130_home_exit();
	enable_endstops(false);
  current_position[Z_AXIS] = 0;
	plan_set_position(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS]); //Set Z axis position equal zero(0). 
	set_destination_to_current();
	destination[Z_AXIS] += 10 * axis_up_dir; //0+10=10mm.
	feedrate = homing_feedrate[Z_AXIS] / 2;
	plan_buffer_line(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS], destination[E_AXIS], feedrate / 60, active_extruder); //Move Z axis continue move up 10 mm. 
	st_synchronize();
	enable_endstops(endstops_enabled);
	if (PRINTER_TYPE == PRINTER_MK3) {
		current_position[Z_AXIS] = Z_MAX_POS + 2.0;
	}
	else {
		current_position[Z_AXIS] = Z_MAX_POS + 9.0; //209+9=218 mm.
	}
    plan_set_position(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS]); //Set max endstop for z axis 218 mm.
	return true;
}
#endif //TMC2130

#ifdef TMC2130
void homeaxis(int axis, uint8_t cnt, uint8_t* pstep)
#else
void homeaxis(int axis, uint8_t cnt)
#endif //TMC2130
{
	bool endstops_enabled  = enable_endstops(true); //RP: endstops should be allways enabled durring homing
#define HOMEAXIS_DO(LETTER) \
((LETTER##_MIN_PIN > -1 && LETTER##_HOME_DIR==-1) || (LETTER##_MAX_PIN > -1 && LETTER##_HOME_DIR==1))
    if ((axis==X_AXIS)?HOMEAXIS_DO(X):(axis==Y_AXIS)?HOMEAXIS_DO(Y):0) //If axis equal X_AXIS do check endstop and if not,check axis equal Y_AXIS do check endstop, and if not homeaxis is 0
	{
        int axis_home_dir = home_dir(axis); 
        feedrate = homing_feedrate[axis];

#ifdef TMC2130
    	tmc2130_home_enter(X_AXIS_MASK << axis); 
#endif //TMC2130

        // Move away a bit, so that the print head does not touch the end position,
        // and the following movement to endstop has a chance to achieve the required velocity
        // for the stall guard to work.
[... 92 lines cut ...]
			lcd_set_cursor(0, 2);
			lcd_print(1);
			lcd_puts_P(_T(MSG_FIND_BED_OFFSET_AND_SKEW_LINE2));
}
			
		bool endstops_enabled  = enable_endstops(false);
        current_position[Z_AXIS] -= 1; //move 1mm down with disabled endstop
        plan_buffer_line_curposXYZE(homing_feedrate[Z_AXIS] / 40, active_extruder);
        st_synchronize();

		// Move the print head close to the bed.
		current_position[Z_AXIS] = MESH_HOME_Z_SEARCH;

		enable_endstops(true);
#ifdef TMC2130
		**tmc2130_home_enter(Z_AXIS_MASK); //Change to tmc2130_home_enter(Z_AXIS)**
#endif //TMC2130

		plan_buffer_line_curposXYZE(homing_feedrate[Z_AXIS] / 40, active_extruder);

		st_synchronize();

So the result after editing firmware as above I can pass all selftests XY axes correct home calibration and XYZ calibration was done. By the way when start final process selftest "first layer calibration" after loaded the filament to extruder and LCD show massage will run first layer calibration, then LCD show error look like massage aways blinking and try press knob many times until LCD show "TMC DRIVER OVERTEMP" then I press the reset button and continue first layer calibration I aways found the error massage "TMC DRIVER OVERTEMP" eventhogh I skip this process and can be calibration first layer from seleting menu, but when I try to print some sample parts like a sqaure box 20x20 mm. the printer can process only a few minute and then LCD show "TMC DRIVER OVERTEMP" again. 
By the way, if I print sample that only a layer like sample level test 200x200 mm it can finish print without any error. So I wonder the error is come from bug that I edit firmware or it come from hardware. Could someone help me solve this issue and if any information that you need please inform me. 
Thank you in advance.

floridaservices, 2020-04-14:
There's nothing wrong with building a clone.  Lots of people do it.  You are asking a lot to expect Prusa to invest their time and resources to support firmware issues with your clone though.

sharkfish2526, 2020-04-14:
@floridaservices 
Thank you for reply. I'm pretty sure all components correct assembled, I tried to check them many time s and do assembly process follow original prusa. So I think there have nothing wrong, however, as I informed if edit the firmware in Marlin_main.cpp I can pass all selftest and calibration except first layer calibration in selftest process will found "TMC DRIVER OVERTEMP" and I found this error many times during print parts. I have check temp of TMC driver on the einsy board, and the result do not found any heat on TMC driver I can touch it by my finger and didn't feel hot. So I think there are bug from firmware. If there's not bug from firmware please help me solve this issue.

floridaservices, 2020-04-14:
No one here is going to be able to help you.  Again, you are asking for firmware assistance for supported firmware installed on unsupported hardware.  So your clone may be a very good replica of what Prusa sells, or it may just be a best approximation from off the shelf parts.  The latter is most likely since you bought it as a kit and did not gather the parts yourself.  Unless every electrical component has very identical specs to what Prusa uses (not likely), there is every indication that this is completely a hardware issue.  Additionally, no one else seems to have this specific problem currently other than you.  The developers are active in the firmware every single day so it's unlikely that this post has gone unnoticed.  You are on your own unfortunately.   

That said, try an older firmware.  Who knows, you might have better luck. with it; time to start learning code if you are going to get anywhere with this.

sharkfish2526, 2020-04-15:
@floridaservices 
I'm so sorry it might be asking wrong place, I don't know this place is support for origin prusa.
By the way, I'm very thankfully for giving me some suggestion eventhough I used clone model.
For the reason why I came to ask question in this place, because I have saw this place gave many good suggestion and solution to people who are have the problem. So I hope some people in this place can support or give me some idea. However, that all rigth and thank you again for your answer. (PS. I have tried older firmware version since 3.7.0, 3.7.1, 3.8.0, 3.8.1, and 3.9.0 RC there are the same result)

closed 2020-04-15 by sharkfish2526

DRracer, 2020-04-15:
Dear @sharkfish2526 , you came asking very politely and provided much information. However, we at Prusa Research don't have the resources to support and help debugging 3D printers cloned from our products. We'd be happy to hear from you if you really solve the problem or find a bug, but judging from the info there is something mechanically wrong on your 3D printer. TMC overheat doesn't necessarily mean the chip is too hot, it can also mean temporary overload current going into the motors. That would also make sense with the problems you are reporting. Please double check the parameters of your motors and/or try using the original ones. Good luck.

@floridaservices thank you for responding faster than we could ;) .