prusa3d/Prusa-Firmware #2605: [BUG] Autoload procedure can trigger during mintemp

awenelo, 2020-04-14:
**Printer type** - MK3S
**Printer firmware version**-  3.8.1

**Describe the bug**
During a Mintemp error, if the filament sensor newly detects the presence of filament (filament sensors state goes from 0 to 1) the Autoload preheat menu will still open.

**To Reproduce**
Unload filament
Turn on Autoload
Unplug hotend thermistor
Insert filament
Autoload preheat menu opens
 
***Or***
Instead of unplugging the thermistor, place in an environment cold enough to trigger mintemp.

**Expected behaviour**
During Mintemp, a preheat menu should not open.

leptun, 2020-04-14:
@awenelo Thank you for reporting this. We'll take a look into the issue and fix it. Please note that this fix might not be part of 3.9.0 as we are nearing the release of it.

I'm thinking that we could disable autoload when any thermal error occurs. Since the user can't realistically do anything with the printer once the error is triggered, this shouldn't cause us problems since we never have to enable it again until reboot.

awenelo, 2020-04-15:
@leptun I've gotten a working fix for it. Should I do a pull request?

leptun, 2020-04-16:
Yes.

closed 2023-10-24 by github-actions[bot]