prusa3d/Prusa-Firmware #2608: [ENHANCEMENT]<PAUSE print after receiving a "THERMAL RUNAWAY" error - MK3S, others

Tanischacat2009, 2020-04-15:
It would be great if a print could just be PAUSED after a THERMAL RUNAWAY error versus having to reset the machine and start from scratch once the issue has been resolved.

I had this happen yesterday during a very long print.  It turned out that the error was caused by the hotend thermistor wire not being firmly connected to the EINSY board on my MK3S printer.  Simply opening up the box and pushing the wire in solved the issue.  I would have then been able to resume my print after a less than 5 minute interruption versus losing a day and a half's worth of time and half a roll of filament due to the failed print.  I do realize that this error can be caused by more serious issues which would require more time and expense to repair.

This feature would be nice to have on any printer, and if for some reason, the user isn't able to or doesn't properly address the cause for the THERMAL RUNAWAY error, this vitally important safety feature would still override one's ability to print until the printer is properly repaired.

floridaservices, 2020-04-15:
It sounds like your thermal runaway protection did what it was supposed to do.  Pause would introduce a new possible point of failure to a vital safety system and would leave the printer "online" during a failure, which could be dangerous.  I understand the frustration of losing a prints worth of filament for nothing, and I hope you are just venting.

Tanischacat2009, 2020-04-15:
Sure, that makes sense, but other functions are available on the menu, including being able to start another print, while the THERMAL RUNAWAY message is displayed, so one would think that a resume option could be available here too.  Obviously, this safety feature would disable heating of any sort of the hotend or the bed until the cause is corrected, even after resetting the printer (which would also cancel any print in progress if not cancelled already).  So, if a print was just paused, it could only be continued IF the thermal runaway issue was properly connected, if not, the error message would either continue to appear or reappear again (like mintemp/mintemp fixed does) when a wire is broken or not connected properly for instance.

landrunner22002, 2020-05-23:
In may case, during printing PC-MAX (polycarbonate) need a bed temperature at 110degreesC and I forgot to close my windows in room: the bed couldn't ensure heat and make a "THERMAL RUNAWAY" but prusa MK2.5S didn't stop or pause printing but stop extruder heating and make jamming with filament... So the idea of Tanischacat2009 is very good !!!

Taylorjtt, 2020-07-02:
I came here to post about this. Twice now I have broken a bed thermocouple wire and a multiday print has been ruined by MINTEMP. 

It would be awesome if the printer generated a new .gcode file that would continue where the print left off and saved it on the SD card so that you could run it once the MINTEMP problem was solved.

I would be able to write something like this if someone could point me to the correct spot in the code.

closed 2023-10-24 by github-actions[bot]