prusa3d/Prusa-Firmware #2611: [BUG]Print speed changes by 1% after changing the speed

espr14, 2020-04-17:
i3 MK3S FW 3.9.0 RC1

1. Start print
2. Increase or decrease the speed by turning the wheel during the print
3. Once you stop, the speed changes after 1 second by 1% up or down (depending on the direction of user change).

For example, I go up to 200% and after a second it will show 201%.

leptun, 2020-04-17:
Fixed by #2585

Photogad, 2020-04-29:
It's not just in the printing speed, it's everywhere in the whole menu. I think the LCD on the mk3 is just too laggy. Lots of times I'll be scrolling with the wheel to the print I want to start  on the SD card, and I think I have the right one selected, but just as I click it to start the print, it selects the wrong one and starts printing the wrong thing.

I've since learned to operate the LCD screen much slower and wait a few seconds before clicking anything.

espr14, 2020-04-29:
The LCD is slow because board is slow. There are lot of compromises so it's hard to do it better. But take a look on Original Prusa Mini. It uses faster board with more RAM. After initialization the LCD is much quicker.

closed 2020-04-29 by espr14

Photogad, 2020-04-29:
> The LCD is slow because board is slow. There are lot of compromises so it's hard to do it better. But take a look on Original Prusa Mini. It uses faster board with more RAM. After initialization the LCD is much quicker.

Any chance well ever see a new board/lcd as an addon to the mk3?  Maybe mk3.5 .... Hmm

espr14, 2020-04-29:
There's a chance, always. But hard to say when. Still, you can take HW from MINI, put it in MK3 and modify the MINI FW (larger print volume). However, MINI board provides single Z driver and there could be other issues.

leptun, 2020-04-29:
> It's not just in the printing speed, it's everywhere in the whole menu. I think the LCD on the mk3 is just too laggy. Lots of times I'll be scrolling with the wheel to the print I want to start on [...]

The Print from SD menu is notorious for this as well. I have a PR that greatly increases the performance in that menu. You could give it a try. #2405

Photogad, 2020-04-29:
> The Print from SD menu is notorious for this as well. I have a PR that greatly increases the performance in that menu. You could give it a try. #2405

will that make it into an official release? I'm stil learning how to compile firmware

leptun, 2020-04-29:
Maybe, the soonest might be 3.10 tho. Enough PRs for now.