prusa3d/Prusa-Firmware #2613: [BUG]No check of filament sensor after loading filament

espr14, 2020-04-17:
i3 MK3S FW 3.9.0 RC1

If I trigger loading of filament, it will try to load the filament and then it asks if it was loaded or not. But it does not check the filament sensor even it is turned on !!! I say that everything is OK and then it e.g. returns to printing to find out just before printing that there's no filament.

After loading it could check the filament sensor (FS) and in case of 0 (FS) it should tell user that filament was not loaded so they should try again or check FS.

awenelo, 2020-04-20:
I believe that this was done to reduce the complexity of the code in the autoload area and is the question "Is the filament extruding and with the right color?" is intended to have the human do exactly that. However, I agree that the filament sensor should be used to check that filament has been loaded.

espr14, 2020-04-20:
It is matter of adding check of FS (one command) and then 2 simple dialogs. If no filament and in idle mode, ask user whether to retry or not. In print mode (filament run out, not pause) just wait for click to retry.

awenelo, 2020-04-21:
We also need to ask if the filament is inserted, as the filament sensor could be broken and reporting a false value.

The simplest possible logic flow that I could come up with is 31 lines, versus the 5 line logic flow from before. Even if every line represented a single line in C, it would still be over 6x the length of the original code.

```
Is there filament? || Is the filament sensor disabled?
Yes:
    'Is it extruding and with the right color?'
    Yes:
        Exit
    No:
        Re-extrude
        Loop back to start
No:
    'Is there filament inserted?'
    Yes:
        'We have an error with the filament sensor'
        Disable filament sensor
        'Filament sensor has been disabled, filament runout detection will no longer work.'
        Exit and resume print
    No:
        Are we printing?
        Yes:
             'Please insert filament or click to load filament'
             Filament inserted || Click:
                 Load filament
                 Loop back to start
        No:
            'Would you like to try to load again?'
            Yes:
                 'Please insert filament or click to load filament'
                 Insert filament || Click:
                     Load filament
                     Loop back to start
            No:
                 Exit
```

espr14, 2020-04-22:
Testing of failure of FS was not there so there's no reason to add it now. The pseudo code would look like this:
```
Is the filament sensor disabled?
Yes:
    follow the original code
No:
    Is there filament? 
    Yes:
        follow the original code
    No:
        "There's no filament. Start the loading by inserting filament again."
        loop back
```

awenelo, 2020-04-22:
Now that we are taking inputs other than the user, we also need to check that the other inputs are correct. For example, if the filament sensor breaks during a print, the printer would have to be reset, as there is no way to exit the `"There's no filament. Start the loading by inserting filament again"` screen. This would leave the user with a half finished print.

Because of this, the first interaction with the user in this logic flow can't be instructing them to load more filament. The first interaction would have to be a question about whether what the filament sensor is reading is wrong. Also, if the filament sensor is broken in the `0` state, which we would now know, we would not be able to start/continue a print, so we have to disable it.

 So, the pseudo-code would have to be:
```
Is the filament sensor disabled?
Yes:
    follow the original code
No:
    Is there filament?
    Yes:
        follow the original code
    No:
        "Is filament inserted?"
        Yes:
            "Faulty filament sensor. I've disabled the filament sensor."
            Disable F. Sensor
        No:
            'Would you like to try to load again?'
            Yes:
                 'Please insert filament or click to load filament'
                 Insert filament || Click:
                     Load filament
                     Loop back to start
            No:
                 Exit
```
However, if we are printing, we know we can't print with no filament, so we also have to check for if we are printing asking the user if they want to load filament. So, we need to replace the `"Would you like to try to load again?"` with `Is variable reload true?` and add this before:
```
Are we printing?
Yes:
    reload = true
No:
    "Would you like to try to load again?"
    Yes:
        reload = true
    No:
        reload = false
```

espr14, 2020-04-22:
The only way how to disable FS (after filament run out) is to load the filament and before the head gets to its last position, press pause and turn FS off. This is not a trivial task.

So better way would be to provide direct switch from filament change to pause mode (if user wants). There you can turn FS off, move axes, do some repairing etc. In such way you don't have to implement disabling of FS during filament change and you add more functionality. This also solves the clogged nozzle issue.

awenelo, 2020-04-22:
If during a print, the filament sensor wires break, it will trigger a filament runout. In a filament runout, the loading procedure is triggered. If we do not disable the FS for the user, thereby forcing the loading procedure to use the old code, we will constantly be looping in the loading state as the filament sensor believes that there is no filament, even when there is filament. So, no matter what we have to disable the filament sensor. However, we could also ask if the user wanted to pause:

```
"Is filament inserted?"
Yes:
    Disable F. Sensor
    "We have a faulty filament sensor. It is now disabled."
    Are we printing?
    Yes:
        "Would you like to pause?"
        Yes:
             Pause
             Exit
        No:
            Exit
    No:
        Exit
```
This would replace the current logic at `"Is there filament?"`

Prusa-Support, 2022-05-12:
Thank you for the report.

Probably the question being asked after a change of filament is "if the color is clear".
This doesn't mean that the IR filament sensor doesn't work as intended.

The IR filament sensor's role is to detect a filament outage, or trigger an autoload routine when a new filament is loaded. 
The verification that the filament is extruded correctly is a task for the user: the printer needs a human to confim that the filament is extruded and exturded with the right color. 

The IR sensor can detect filament before the extruder gears (or in the extruder gear for MMU2S), but it doesn't know what happens "after the extruder".
A jam could prevent the filament from being extruded, or - because you may load a different filament color - the printer needs the user to check if there is "color bleeding". 

The IR filament sensor detects filament (usually) before the extruder gear, but is not capable of detecting a jam or color bleeding.
I hope this makes sense.
In other words, this is not a bug but the intended behavior.

Maybe the message could be rephrased to be more clear?
Please share your thoughts.

Michele Moramarco
Prusa Research

espr14, 2022-05-12:
I don't have any i3 MK3 any more so it's not relevant for me any more. You can close the issue.

Prusa-Support, 2022-05-13:
Thank you for the update anyway.
We may consider rephrasing the specific message based on your feedback.
Your feedback remains most appreciated.

Michele Moramarco
Prusa Research

closed 2022-05-13 by Prusa-Support