prusa3d/Prusa-Firmware #2614: [ENHANCEMENT]What if nozzle is clogged during filament loading...

espr14, 2020-04-17:
i3 MK3S FW 3.9.0 RC1

I'm loading filament. After automatic loading it asks
1. OK?
2. Correct color?
3. Not loaded?

Then it behaves:
1. continues with printing
2. pushes filament a little
3. pushes filament a lot

But what if the nozzle is clogged? Filament is not going in any more and does go out (wheel hold it). There's no option to unload the filament and pause the print. The only option is to select OK and then go to menu and press pause before it starts to print.

Panayiotis-git, 2020-04-17:
After the Load/Autoload filament functions complete, they ask the following question:
```
Filament extruding &
with correct color?
>Yes
 No
```

So, I assume that you are referring to the M600 (change filament) functionality during a print.
If so, have a look at the [LCD option to unload new filament on M600 (issue #2415)](https://github.com/prusa3d/Prusa-Firmware/pull/2423)

espr14, 2020-04-17:
Duplicate #2415.

closed 2020-04-17 by espr14