prusa3d/Prusa-Firmware #2616: [ENHANCEMENT]Multiple Z lifts

espr14, 2020-04-17:
i3 MK3S FW 3.9.0 RC1

Before mesh bed level, printer does not know its height. Therefore, there are Z lifts at these situations:
1. nozzle heat is started
2. unload of filament is triggered
3. load of filament is triggered

If all occurs before bed leveling, there are 3 lifts. It is not necessary, there could be single lift instead.

leptun, 2020-04-17:
All of the functions above have a minimum Z distance for execution. If the nozzle is above that distance, no move is done, but if it’s lower then it will raise to that minimum z position.
Here is a snippet with the minimum positions. You may notice how executing them in the order you wrote above, you’d first raise to 10mm, then to 20mm and in the end to 50mm.

https://github.com/prusa3d/Prusa-Firmware/blob/c07814bcad3570c5f2b89cabb0e07fe0ab4cea19/Firmware/Configuration.h#L554-L556

espr14, 2020-04-17:
Great. Still, you can use same value for preheat and unload.

closed 2023-10-19 by github-actions[bot]