prusa3d/Prusa-Firmware #2618: [ENHANCEMENT] Sort SD card files every time clicking "Print from SD Card" if FlashAir is used

Saij, 2020-04-18:
**Printer type** MK3S
**MMU Upgrade** none

**Is your enhancement related to a problem? Please describe.**
No
  
**Describe the solution you'd like**
As I use a FlashAir card to wirelessly transfer G-Code to my printer the SD Card never gets removed. So the files don't get sorted when I click on "Print from SD" and that is a little thing that bugs me. I would suggest to sort files every time you click on "Print from SD" if card type is FlashAir.

leptun, 2020-04-18:
https://github.com/prusa3d/Prusa-Firmware/pull/2405
This is one of my PRs. In it I do a complete overhaul of the Print from SD menu. It has faster sorting and it also lags much less when scrolling through files.
I haven't noted it anywhere, but I also included a new button in the menu called ":arrows_counterclockwise:Refresh". It basically resorts the files in the current folder. It is enabled only if you switch to FlashAir mode.

gogglespisano, 2020-06-28:
I don't know if it needs to re-sort every time you use Print from SD, but it would be nice if it did whenever there was a file change.  

It will show a newly added file, so it could use the same detection to set a flag and do a sort the next time Print from SD is selected.

closed 2023-10-19 by github-actions[bot]