prusa3d/Prusa-Firmware #2620: [ENHANCEMENT]Main menu is shifted

espr14, 2020-04-18:
i3 MK3S FW 3.9.0 RC1

Go to menu "Support" and click on "Date:". It will return you to the main menu but the "Support" line is not at the bottom even it is the last item in the list.

leptun, 2020-04-18:
The issue is that the "view" is not saved when switching menus, just the selected item. As a result, the view is kept from the previous screen. The support menu is the only one which has a "back" item so low in the menu. All other menus have the "back" button at the top. We could do a check, where when changing menus, the view is adjusted automatically to the limit.

MohanMaker, 2020-09-07:
Same issue here with firmware version 3.9.0 - 3421 on MK3S. Do you have a fix @leptun? Maybe it could be implemented into 3.9.1?

leptun, 2020-09-07:
@MohanMaker Thank you for reminding me of this issue. I just wrote a quick patch that should fix the issue described by OP. Not sure if there is enough time for it to be merged into the 3.9.1 final release, but in 3.10 for sure. 
If you want to give the patch a try, the PR with the fix is #2828

MohanMaker, 2020-09-07:
Great, will give it a try. Excited that it will be in 3.10!

PR #2828 (2020-09-08): Menu view return patch
Fixes #2620
closed 2020-09-08 by DRracer