prusa3d/Prusa-Firmware #2624: [ENHANCEMENT] Abort octoprint job at the printer directly

Pace17881, 2020-04-21:
**Printer type**
MK3S
**MMU Upgrade**
no

I want to abort a print job directly on the printer which was started by octoprint in case the print is not as expected. I need an option in the printer menu to abort a print.

PaulGameDev, 2020-04-21:
Could possibly be implemented in software, but as long as it isn't, it's also really easy to just press the X button (abort) and move the Extruder out of the way manually, when you're already wanting to operate the lcd, you are probably in front of the printer, right? ^^

Pace17881, 2020-04-21:
Pressing the x button is a complete reset of the printer and not the expected behavior.

aequitas, 2020-11-01:
I'm using a Raspberry pi zero w connected to the Einsy board for octoprinting and missing this feature as well. Since the UI is really slow on this underpowered Pi I tend not to use it and also to keep it from consuming resources that should go to sending print commands. So controlling Octoprint from the LCD and knop would be a highly appreciated feature.

closed 2023-10-19 by github-actions[bot]