prusa3d/Prusa-Firmware #2626: [BUG] Firmware crashes to random characters on screen when mintemp is triggered

jimhigson, 2020-04-22:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK2.5S
**Printer firmware version**-  3.8.1

**MMU Upgrade** - no
**Describe the bug**
Please see [this thread in the forum](https://forum.prusaprinters.org/forum/original-prusa-i3-mk2.5s-mk2.5-hardware-firmware-and-software-help/printer-screen-crashes-displays-random-gibberish-chars-mid-print/) for a screenshot

There is a bug in the firmware whereby the printer screen instead of properly reporting a mintemp error, it shows random chars. This makes it very difficult to diagnose mintemp issues without an octopi since there is no reporting of what the problem is.

We can see that this is a mintemp from the console in octopi.
  
**To Reproduce**
Start a print, trigger a mintemp in some way. In my case this was caused by a faulty thermistor. Probably unplugging the thermistor mid-print would work.
  
**Expected behavior**
On causing the mintemp, random chars will be shown on the screen rather than an error message.
  
**G-code**
Applicable to any g-code.

**Video**
  Please attach a video. It usually helps to solve the problem.

leptun, 2020-04-22:
Already fixed in 3.9-RC1

3d-gussner, 2020-04-23:
@jimhigson Please consider to close this issue as this has been fixed already.

closed 2020-05-13 by JohnnyDeer