prusa3d/Prusa-Firmware #2628: [BUG] Firmware 3.8.1 crashing when sent M112

gchedzhemov, 2020-04-23:
**Printer type** - [MK3S]
**Printer firmware version**-  [3.8.1]

**MMU Upgrade** - [MMU2S currently, but also happens without MMU2S]
**MMU upgrade firmware version** [1.0.6]

**Describe the bug**
Sending M112 to a Prusa printer running firmware 3.8.1 causes an immediate firmware crash with a garbled screen. End state is identical to that reported in [Issue 2343](https://github.com/prusa3d/Prusa-Firmware/issues/2343)
 
**To Reproduce**
Send M112

**Expected behavior**
Full Emergency Stop as described in [M112 section of the ReRap wiki](https://reprap.org/wiki/G-code#M112:_Full_.28Emergency.29_Stop): 

> Any moves in progress are immediately terminated, then RepRap shuts down. All motors and heaters are turned off. It can be started again by pressing the reset button on the master microcontroller. See also M0 and M1. 

The printer does not display garbled output.

This is valuable as software like the Emergency Stop plugin for Octoprint relies on M112 to work.
  
**G-code**
N/A. Just issue M112 from a terminal

3d-gussner, 2020-04-23:
@gchedzhemov Good morning, thanks for the bug report and detailed information you added.
I am also glad to see that you already searched the issue queue before opening a new bug report, even you might have missed a note 😉 .

This has been fixed as mentioned in https://github.com/prusa3d/Prusa-Firmware/issues/2343#issuecomment-561016748 by https://github.com/prusa3d/Prusa-Firmware/pull/2348

I tested it few minutes ago by starting a small print via Pronterface:
- Let it heat up
- Print the purge line
- During skirt print I run the M112 g-code
- It sopped not immediately due to a long line, it needed to finish the current g-code line (which is normal)

On the LCD screen I can see:
M112 called.
Emergency stop.

- The Heated Bed LED immediately went off
- The printer doesn't react to any additional g-codes as it should be see Marlin documentation "A reset is required to return to operational mode."
- I can move the X/Y/Z axis by hand that means that the steppers are disabled as well.
- Reset the printer after few minutes and the current Nozzle and Bed temperature was way lower than the print temp, so also the hotend heater stopped also with M112.

If you want to test it yourself please compile the latest version from this branch.

Please also to consider to close this issue as it has been solved.

gchedzhemov, 2020-04-23:
Confirmed resolved with 3.9.0 RC1.

Issuing M112 now results in "M112 called. Emergency Stop", as expected. Thank you for the prompt reply @3d-gussner, and for pointing me to the appropriate resolution!

closed 2020-04-23 by gchedzhemov