prusa3d/Prusa-Firmware #2633: Warn user about improper GCODE temps

Lee-Hickman, 2020-04-25:
I had an issue where my daughter tried to print the Prusa Buddy (meant for PLA), but the printer had PETG filament loaded. She hit print and waled away. She came back 30 minutes later to a glob of plastic stuck to the nozzle.

My enhancement is this: The printer knows what filament is loaded, because it does not ask me what temperature to preheat to when unloading filament. The printer should do a "self check" before beginning to print the gcode to ensure that the print temperatures are within range for the loaded filament. If not, put a warning on the screen that the "Temperature for the filament loaded is not within range" and allow the user to move on if they need a not standard temperature, or the option to load the correct filament.

Bonus points: Allow the user to adjust the min/max temp for each filament in the settings.

3d-gussner, 2020-04-26:
@Lee-Hickman Thanks to the report and I am sorry that you got a glob of plastic.

The printer doesn't know what filament is loaded.
- It knows if the filament sensor is active that "some" filament is loaded. So it starts to heat up and print.

At this moment I don't know if we could implement something so the printer "knows" what filament is loaded.

Here some thoughts
- We may be able to save the last print and bed temperatures, BUT
  - this would need to be written to EEPROM after every print and that isn't good the the chip
  - Let's say we do it:
    - The printer "knows" what the last print temp was and could popup a message and ask for interaction if these values are way off with new gcode temps
       - But what if you changed the filament manually? How do we handle that
...
You see there are lot of ifs and thens and we probably miss something.

If you have could provide a if then process it may help what may be possible what not.

Lee-Hickman, 2020-04-26:
I currently only have access to a Prusa MINI, so my information will be scoped to that machine.

When I select "Load Filament", I select a type of material to load (PLA). The printer then heats to the appropriate temperature (215) to do its thing. I can now select "Unload Filament" and the printer knows exactly what temperature to reach (215) before it starts unloading the filament. This tells me this information is stored somewhere on the Prusa MINI. If you are telling me I'm wrong, that's OK. The enhancement will start there. The way I see it, the enhancement should look like:

Part 1 (Saving the filament type): 
- Save the selection from the LOAD FILAMENT selection.
- Upon a selection of UNLOAD FILAMENT clear the entry

Part 2 (Comparing loaded filament vs. gcode):
- [Option 1] Only when initializing the print (at the very beginning) does the printer do a check against current loaded filament type and requested temperature in the gcode. 
- [Option 2] I also know I select the filament type in Prusa Slicer (I am not too familiar with others to know if this is true there). Before the print begins there could be a check to compare the 2 setting (loaded filament and gcode from Prusa Slicer) to determine if the filaments match
- Either way this check would only be performed at the beginning of the print right before the first bit of plastic is extruded.
- If check succeeds. Nothing changes. Print as usual.

Part 3 (Check fails):
- An on-screen prompt alerts the user user that the filament check has failed.
- The user is given options to handle the failed check.
    - [Only need if doing temp range logic] OK through the check (loaded filament matches the desired filament for printing and the user is using outside of range temperatures. Proceed normally
    - Option to cancel the print in order to change filament
        - The user realizes they are trying to print a PLA gcode with PETG loaded in the printer
    - Option to set the current Loaded Filament to match the gcode
        - The user has done a manual change and the printer does not know. This allows for correction to the correct filament type

3d-gussner, 2020-04-28:
@Lee-Hickman Thanks for the update and sorry for the late answer.
Just a short question: Is the inital related to the Prusa Mini or MK2.5/s MK3/s?

We normally have an template but it seams that you didn't get this or you deleted the needed information. How did you create this issue, as if i try to open an new one there are 4 options. 

If this issue is Prusa Mini related then you should post this https://github.com/prusa3d/Prusa-Firmware-Buddy/issues

Lee-Hickman, 2020-04-28:
This is for the MINI. I figure, if it can be done for other models of the printer, this would be a feature that I think would be valuable no matter the platform. But my primary printer is the MINI.

I was given the link from the Chat Support and clicked the link I was provided. I performed a search to see if there were any issues already submitted. If I follow the same steps, I am prompted, like you mention, to fill out a form. But I did not at the time of creation. 

I guess I'll recreate this issue in the other github repository.

3d-gussner, 2020-05-01:
@Lee-Hickman Saw that you opened the issue in the Mini repository, hope you get help there.

We will see if that is an option for other Prusa printers (noted), thanks again for the feedback.

If you don't mind can you please close this issue here?

Lee-Hickman, 2020-05-01:
This issue was in the wrong repository.

closed 2020-05-01 by Lee-Hickman