prusa3d/Prusa-Firmware #2634: [BUG]<3.8.1 RC1 2851 not working into 3.9.0 RC! ? RE: Z-Leveling Enforcemetn

SRIFRAF, 2020-04-26:
**MK2.5S** 
- 3.9.0 RC1  Still had Z-Leveling enforcement Fail for me
-3.8.1 RC1 2851 "Partial" working

If I posted this in the wrong area, please forgive me, first time posting...
I recently upgraded my experienced Mk2S to a full Mk2.5S, I am experienced in building and the upgrade went well and everything seemed to perfect spec. After upgrading the firmware to the latest Prusa 3.8.1 after the initial perfect XYZ calibration on the Z-height calibration I started getting Z-leveling enforcement problems and failures.  I tried several firmware's and nothing worked including the new 3.9.0 RC1 that I read had the issue corrected and anticipated it working.  Of course on each try I did a factory reset, firmware flash, factory reset, firmware flash...  I did install the 3.8.1 RC1 2851 build and could get the printer to finally start and finish a print without the Z-Leveling Enforcement fail, the Benchy is prefect, EXCEPT now when repeating the print the first layer z height is all over the place, either too high or right down to gouging out my bed (ouch), I can't adjust the live Z fast enough to get it right.  
Is there a fix to this?, perhaps not automatically enabled in 3.9.0 RC1 ?
I am still at 3.8.1 RC1 2851 build for now as it seems the only way to get rid of the Z-leveling error.

SRIFRAF, 2020-04-27:
In other words the fix in 3.9.0 RC1 was supposed to turn off the heatbed PWM whenever the Pinda probe was activated as this caused interference and could trigger a Z-Leveling Enforcement failure at the start of a leveling procedure, something done at the start of every print.  Seems this firmware did not fix it and it is still failing.  The 3.8.1 RC1 2851 addressed this issue and allowed the mesh leveling to complete so the print could start... and as I understand it was supposed to be carried into 3.9.0 RC1.... but the issue now is it creates a problem with the probe implementing the correct initial layer Z height and messes that up, so your nozzle at the start will be way too high or way too low and damage can result. Hope that makes sense...

SRIFRAF, 2020-04-28:
My great original Prusa Mk2S that has been an absolute solid printer for so long has now basically been sent into retirement after this Mk2.5S upgrade, the varying Z heights at the start are a real issue that are causing big time scratches on my steel plates, Anyone have any other suggestions?

3d-gussner, 2020-05-03:
@SRIFRAF Please try RC3 as there have been made some improvement especially for the MK2.5s

SRIFRAF, 2020-05-03:
Thanks for the comments… I put this below under: 2641, tried the RC3 right after it came out, I had already posted this bug...
"Flash, restore, Flash to 3.9.0 RC3 went fine but problem remains..
 I upgraded from 3.8.1 RC1 2851, that was the last FW build that solved my Z-Leveling Enforcement when starting a mesh for a print but I was still having initial Z-height inconsistences to the point of scratching the bed... This upgrade to 3.9.0 RC3 brings back the Z-leveling Enforcement problem during the Mesh level before every print again. Both XYZ Cal and machine is perfect, seems like this relationship with the probe and bed temp is an unending painful issue?"

SRIFRAF, 2020-05-22:
The 3.9.0 new download posted 5/18/22 direct from Prusa solved the issue

closed 2020-05-22 by SRIFRAF