prusa3d/Prusa-Firmware #2635: [FEATURE REQUEST]Store Z calibration per material

PatrickSannes, 2020-04-27:
At this moment it is possible to calibrate the first layer zheight per material. This is very nice to tune in ABS or PETG. 
Thing is, for some reason I run into different heights per material (ABS is heigher, PLA is lower). It would be nice to store these inside the firmware per material. In this way I dont have to keep notes.

Patrick

awenelo, 2020-04-27:
Instead of keeping notes, you could set different steel sheet profiles for each one and switch between them. The filament chosen for the first layer calibration only affects the preheat temperatures and is not stored after the preheat temperature is set. To provide the requested functionality, we would need to store up to 56 first layer calibration values, compared to the 8 before. (7 different filament options, 8 steel sheet profiles, 7*8=56)

leptun, 2020-04-27:
@PatrickSannes Also, there’s a calibration called “Temperature Cal.”. It compensates for this exact drift. Have you tried it?

closed 2023-10-19 by github-actions[bot]